rdiankov/openrave #335: CheckSelfCollision always returns false on cloned robot

mkoval, 2015-01-02:
`CheckSelfCollision` seems to be broken on cloned robots: the function returns false. This definitely occurs for the configuration the robot was cloned in and, possibly, elsewhere. This Python snippet replicates the bug:

    robot.SetActiveDOFValues(config_self_collision)
    
    cloned_robot = openravepy.RaveCreateRobot(env, robot.GetXMLId())
    cloned_robot.Clone(robot, 0)
    env.Add(cloned_robot, True)
    
    assert robot.CheckSelfCollision() # succeeds
    assert cloned_robot.CheckSelfCollision() # fails

I attempted to isolate the bug by exploring a few variations of this:

* Cloning the entire environment, e.g. using `CloneSelf`, also triggers the bug.
* Directly calling `env.GetCollisionChecker().CheckSelfCollision(robot, 0)` also triggers the bug.
* Setting the cloned robot's DOF values to an intermediate value, then back to `config_self_collision` also triggers the bug.

This rules out some of the obvious types of caching that could cause a bug like this. This is a **major bug**: we only discovered this because a planner was happily returning a trajectory with the start state in self-collision.

@rdiankov: Could you take a this? The cloning code is quite complicated and I'm out of ideas.

rdiankov, 2015-01-02:
i have one idea: when the robot is loaded initialized, its current configuration is assumed to be always not in self-collision (ie the zero position), therefore any colliding links actually get put in the ignore. I'm guessing that the cloning is somehow accidentally calling this function....

one way around it is to first set your robot dofvalues to all zeros before cloning:

```
with robot:
    robot.SetDOFValues(zeros(robot.GetDOF()))
    cloned_robot = openravepy.RaveCreateRobot(env, robot.GetXMLId())
    cloned_robot.Clone(robot, 0)
    env.Add(cloned_robot, True)
```

nevertheless, i'll take a look at the problem.

mkoval, 2015-01-02:
I thought the same thing, but `GetAdjacentLinks()` returns an identical list on both `robot` and `cloned_robot`.

I also have a two questions about the "non-colliding configuration":
- How do I set this to a non-zero configuration?
  - On HERB, `zeros(robot.GetDOF())` is not a valid configuration: it's outside of joint limits.
- What is the purpose in having a non-colliding configuration at all?
  - There is no reason to think that the zero configuration is not in self-collision.
  - I can easily `SetDOFValues` to a configuration that is in self-collision, so this is a bit of a strange requirement.

mkoval, 2015-01-06:
@rdiankov: Have you had a chance to look at this?

Sorry for bugging you about it, but it's blocking a few of our internal pull requests. I still haven't figured out what the source of the problem is.

DavidB-CMU, 2016-01-14:
@rdiankov Hi Rosen, do you have any suggestions please regarding this issue?