rdiankov/openrave #340: enabled vs. static vs. active for fixed joints

benaxelrod, 2015-01-26:
I have some "fixed" joints in my kinematic chain.  I am using these to connect different parts of the robot for different configurations.  (Different end-effectors for example).  But i found that ikfast didn't like this:

> openrave.py --database inversekinematics --robot=arm_wedge_hand.xml --manipname=bottle --freeinc=0.01 --iktype=Translation3D
> Traceback (most recent call last):
>   File "/usr/local/bin/openrave.py", line 126, in <module>
>     database.run(args=args)
>   File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1104, in run
>     InverseKinematicsModel.RunFromParser(*args,**kwargs)
>   File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1069, in RunFromParser
>     model = DatabaseGenerator.RunFromParser(Model=Model,parser=parser,robotatts=robotatts,args=args,**kwargs)
>   File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/__init__.py", line 252, in RunFromParser
>     model.autogenerate(options=options)
>   File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 644, in autogenerate
>     self.generate(iktype=iktype,freejoints=freejoints,precision=precision,forceikbuild=forceikbuild,outputlang=outputlang,ipython=ipython)
>   File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 867, in generate
>     chaintree = solver.generateIkSolver(baselink=baselink,eelink=eelink,freeindices=self.freeindices,solvefn=solvefn)
>   File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 2083, in generateIkSolver
>     LinksRaw, jointvars = self.forwardKinematicsChain(chainlinks,chainjoints)
>   File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 1567, in forwardKinematicsChain
>     raise ValueError('cannot solve for mechanism when a non-mimic passive joint %s is in chain'%str(joint))
> ValueError: cannot solve for mechanism when a non-mimic passive joint <joint:fixed_wrist_plate_to_wedge (-1), dof=-1, parent=Prototype_A_Wedge_Hand> is in chain

Where my XML looked like this:

```xml
<Joint enable="false" name="fixed_wrist_plate_to_wedge" >
  <Body>wrist_plate</Body>
  <Body>wedge</Body>
</Joint>
```

I guess i (wrongly) assumed that disabling the joint makes it fixed.  Looking a few lines above the throw in ikfast.py, i see that if `joint.IsStatic()` then it will do the right thing.  But what makes a static joint?

Looking at the code for `KinBody::Joint::IsStatic()` in kinbodyjoint.cpp, I see that (ignoring mimic joints) the joint must not be circular and the limits must be equal.  Sure enough, adding `<limitsdeg>0 0</limitsdeg>` to my XML allows ikfast to run properly.  But what does enable do then? 

Looking at xmlreaders-core.cpp, setting the enable flag to false, sets an IsActive flag in the JointInfo class.  Which says that if true it should belong to the DOF of the body.  I am not exactly sure of the implications of this, but that is as far as i traced it.

So it seems to me that there is a bunch of different terms for kind of the same thing, but slightly different.  Each interface has its own terminology.  I don't know the best way to straighten it out, but perhaps IsStatic() should also check for if the joint is enabled / active.

rdiankov, 2015-01-27:
in terms of XML parsing, setting the joint limits to 0, 0 (like you mentioned) is the only way to make it parse as static. There is no static="true"/"false" attribute because of that.

As for "enable", you've probably noticed openrave has active and passive joints. active joints participate in the robot's DOF and can be set while passive joints cannot be set directly. When enable="false", the joint becomes passive and not counted in the robot's dof.

hope this helped!

closed 2015-01-27 by rdiankov

benaxelrod, 2015-01-28:
Thanks for the clarification.  I went back and re-read the [XML spec](http://openrave.programmingvision.com/wiki/index.php/Format:XML) and saw where this behavior is described.