rdiankov/openrave #346: 0.9 fails to compile under Ubuntu 14.04 - no /usr/lib/libboost_thread-mt.so

pirobot, 2015-05-07:
Hello,

I am trying to compile version 0.9 master branch on a new installation of Ubuntu 14.04.  After running cmake, make is bailing with the following error:

[ 14%] Built target check_libm_accuracy
[ 15%] Built target fparser
make[2]: *** No rule to make target `', needed by `src/libopenrave/libopenrave0.9.so.0.9.0'.  Stop.
make[1]: *** [src/libopenrave/CMakeFiles/libopenrave.dir/all] Error 2
make: *** [all] Error 2

I have installed the libboost-all-dev package bu there is no library named libboost_thread-mt.so.

rahulpatil323, 2015-07-02:
-mt is not needed. You can workaround by having a symbolic link to libboost_thread.so.

stephane-caron, 2015-11-25:
+1. I recently built openrave from source on a fresh Ubuntu 14.04 ([full install instructions here](https://scaron.info/teaching/installing-openrave-on-ubuntu-14.04.html)) and I only installed ``libboost-all-dev`` and ``libboost-date-time-dev``. Here is the full lib install line:

```bash
sudo apt-get install libassimp-dev libavcodec-dev libavformat-dev libavformat-dev libboost-all-dev libboost-date-time-dev libbullet-dev libfaac-dev libglew-dev  libgsm1-dev liblapack-dev libmpfr-dev libode-dev libogg-dev libopenscenegraph-dev libpcre3-dev libpcrecpp0 libqhull-dev libqt4-dev libsoqt-dev-common libsoqt4-dev libswscale-dev libswscale-dev libvorbis-dev libx264-dev libxml2-dev libxvidcore-dev
```