rdiankov/openrave #349: multiple free joints and increment value

benaxelrod, 2015-06-10:
It appears that when you have a manipulator chain with multiple free joints, and you want to set a different increment for each, the value you specify for the first will be used for all of them.  This is strange because if you call ```manip.ik.load(freeinc=[0.01])``` it complains if you use the wrong number of values.  (But all except the first are ignored).  

This is a little hard to test.  I only noticed it when i set ```manip.ik.load(freeinc=[0.1, 0.01])``` (because i have 2 free joints), then looked at the results of ```manip.FindIKSolutions()``` and both free joints were in increments of 0.1.

The cause is easily seen in these lines from ```setrobot()``` in inversekinematics.py:
```
if len(self.freeinc) > 0:
    freeincvalue = self.freeinc[0]
```
However, the fix is kind of a pain because a variable number of parameters will have to be sent to ```_SetDefaultIncrementsCommand()``` in ikfastsolver.cpp.  I don't know that much about how this code is supposed to work to fix it.  Would 4 parameters need to be sent for each free joint?

benaxelrod, 2015-06-11:
i implemented the bugfix.  I'll try to submit a pull request tomorrow.

rdiankov, 2015-06-12:
hmm........ SetDefaultIncrements is for a different type of free joint. When two axes align in the IK, it will create a new free variable. that's what it is for...

you can SendCommand('GetFreeIncrements') and SendCommand('SetFreeIncrements ...') to the ik solver to confirm this behavior.

benaxelrod, 2015-06-12:
Well, I don't know about that...  All i did was trace the calls from ```ik.load()``` to ```setrobot()``` to ```SendCommand()``` to ```_SetDefaultIncrementsCommand()```, and it was clearly ignoring all but the first element in the freeinc list that was passed into ```ik.load```.  See my pull request for the patch.