rdiankov/openrave #354: building 3rdparty lib convexdecomposition fails on arm architectures: bad instruction pause

morxa, 2015-07-01:
`NvThreadConfig.cpp` contains the assembler instruction `pause`, which is not a valid assembler instruction on arm architectures:
```
void tc_spinloop()
{
#ifndef _MSC_VER
      asm ( "pause" );
   #else
      __asm { pause };
   #endif
}
```
This causes a build failure for the 3rdparty library convexdecomposition, and thus for openrave:
```
[  6%] Building CXX object 3rdparty/convexdecomposition/CMakeFiles/convexdecomposition.dir/NvThreadConfig.cpp.o
cd /builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/build/3rdparty/convexdecomposition && /usr/bin/c++   -DCLOCK_GETTIME_FOUND -DLINUX -DUSE_EXT_HASH_MAP -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -march=armv7-a -mfpu=vfpv3-d16  -mfloat-abi=hard  -O2 -g -I/builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/cpp-gen-md5 -I/usr/include/libxml2    -fno-strict-aliasing -Wall -fPIC -I/usr/include/libxml2 -fPIC -o CMakeFiles/convexdecomposition.dir/NvThreadConfig.cpp.o -c /builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/3rdparty/convexdecomposition/NvThreadConfig.cpp
[  6%] Linking CXX executable check_libm_accuracy-native
cd /builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/check_libm_accuracy-native.dir/link.txt --verbose=1
/usr/bin/c++   -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -march=armv7-a -mfpu=vfpv3-d16  -mfloat-abi=hard  -O2 -g   -Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld CMakeFiles/check_libm_accuracy-native.dir/check_libm_accuracy_main.cpp.o  -o check_libm_accuracy-native -rdynamic -lcrlibm -lstdc++ 
/builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/3rdparty/ivcon/ivcon.cpp: In function 'bool ivcon::ReadFile(const char*, std::vector<float>&, std::vector<int>&)':
/builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/3rdparty/ivcon/ivcon.cpp:2676:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(size_t i = 0; i < cor3_num; ++i) {
                         ^
/builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/3rdparty/convexdecomposition/NvThreadConfig.cpp: In function 'void CONVEX_DECOMPOSITION::tc_releaseThreadMutex(CONVEX_DECOMPOSITION::ThreadMutex*)':
/builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/3rdparty/convexdecomposition/NvThreadConfig.cpp:289:10: warning: deleting object of polymorphic class type 'CONVEX_DECOMPOSITION::MyThreadMutex' which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor]
   delete m;
          ^
/builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/3rdparty/convexdecomposition/NvThreadConfig.cpp: In function 'void CONVEX_DECOMPOSITION::tc_releaseThreadEvent(CONVEX_DECOMPOSITION::ThreadEvent*)':
/builddir/build/BUILD/openrave-b32979b07426b7f0dc247e5516b57a44cbeaf4a3/3rdparty/convexdecomposition/NvThreadConfig.cpp:474:10: warning: deleting object of polymorphic class type 'CONVEX_DECOMPOSITION::MyThreadEvent' which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor]
   delete m;
          ^
{standard input}: Assembler messages:
{standard input}:640: Error: bad instruction `pause'
```

This was found while packaging openrave for Fedora. I would guess `pause` could be replaced by a suitable instruction for arm, but I'll simply exclude arm from the supported architectures for the package until this is fixed.

rdiankov, 2015-07-01:
wow, i didn't even realize that library was doing that. perhaps we can figure out a way to get rid of the instruction...

morxa, 2017-10-21:
For reference: This also affects s390x.