rdiankov/openrave #355: libopenrave-core calls exit

morxa, 2015-07-01:
The shared library `libopenrave-core.so` calls `exit()`, which is in many cases undesired behavior for a shared library, because it causes the main binary to terminate without any cleanup. 

This was found while creating a Fedora package for openrave. Fedora's packaging guidelines recommend to report such issues to upstream.

I haven't looked at it into detail; but it would probably make sense to replace the `exit()` call with some kind of error handling.

Thanks for your help!

rdiankov, 2015-07-01:
i agree about the exit calls. i just grepped for any exit fn call, but couldn't find any hits. perhaps you can point out where the exit call is so i can remove it.

thanks

morxa, 2015-07-07:
The call to `exit` is in the 3rdparty lib [ivcon](https://github.com/rdiankov/openrave/blob/9c48ee107c8d5103d52940cca5317bb428440993/3rdparty/ivcon/ivcon.cpp) in [line 7512](https://github.com/rdiankov/openrave/blob/9c48ee107c8d5103d52940cca5317bb428440993/3rdparty/ivcon/ivcon.cpp#L7512). Looking at the code it seems like it's a typical divide by zero error. So maybe you could throw an exception instead of calling `exit`?