rdiankov/openrave #358: Make march=native user-definable

languitar, 2015-07-13:
Since the addition of `march=native` to the compile flags, we are unable to build openrave so that different processors are supported. Moreover, due to a qemu bug, compilation even fails on a VM. I can understand why this is generally a desired setting in case you optimize the builds specifically for a machine, but in terms of more generic packaging, this breaks many things. Therefore, please provide a CMake option to make this behavior optional.

rdiankov, 2015-07-14:
good point, just made it so that NATIVE_COMPILE_FLAGS is empty by default

c5e7599b3e302a2aacea43b82f97abf9fd3a04fe

closed 2015-07-14 by rdiankov

rdiankov, 2015-07-14:
just curious, can you tell me what the errors were? it's hard to imagine how march=native can break builds.
thanks

languitar, 2015-07-14:
Thanks. We endet up in exactly this:
http://dpdk.org/ml/archives/dev/2014-June/003748.html

languitar, 2015-07-16:
Could you please apply the fix also to latest_stable?

rdiankov, 2015-07-16:
np, done ;0)