rdiankov/openrave #362: Help with installation (latest_stable branch on ubuntu 14.04)

robotcd, 2015-08-27:
Hello! I'm getting started with openrave and I would appreciate help in getting up to speed.
I installed openrave (latest_stable branch) on ubuntu 14.04 and tried to run openrave.py --example hanoi to test if the basic installation is working (feel free to suggest other test). I got the following error:
Let me know if you know the solution.
Thanks!

$ openrave.py --example hanoi
[plugindatabase.h:929 _SysLoadLibrary] /usr/local/share/openrave-0.9/plugins/openravepy_configurationcache.so: libconfigurationcache.so: cannot open shared object file: No such file or directory
[plugindatabase.h:857 _LoadPlugin] failed to load: /usr/local/share/openrave-0.9/plugins/openravepy_configurationcache.so
[plugindatabase.h:577 Create] Failed to create name ode, interface collisionchecker
[plugindatabase.h:577 Create] Failed to create name bullet, interface collisionchecker
openravepy.databases.convexdecomposition: load, python h5py library not found, will not be able to speedup database access
openravepy.databases.linkstatistics: load, python h5py library not found, will not be able to speedup database access
robot resolutions:  [ 0.00195998  0.00198425  0.00328496  0.01093304  0.0106936   0.0234112
  0.02297028]
robot weights:  [ 1.52329723  0.92804625  0.38978808  0.15792621  0.10813445  0.04291764
  0.02730175]
hanoimove disk2 from srcpeg to destpeg
opening hand
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 149, in <module>
    example.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/openravepy_ext.py", line 64, in newfn
    return fn(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 235, in run
    OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 239, in parseAndCreateThreadedUser
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 101, in SetViewerUserThread
    userfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 239, in <lambda>
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 215, in main
    hanoi.hanoisolve(3,hanoi.srcpeg,hanoi.destpeg,hanoi.peg)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 205, in hanoisolve
    self.hanoisolve(n-1, pegfrom, pegby, pegto)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 205, in hanoisolve
    self.hanoisolve(n-1, pegfrom, pegby, pegto)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 199, in hanoisolve
    if not self.hanoimove(disk, pegfrom, pegto, self.heights[len(pegto.disks)]):
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 166, in hanoimove
    openhandfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 158, in <lambda>
    openhandfn = lambda: self.MoveToPosition([-0.7],self.robot.GetActiveManipulator().GetGripperIndices())
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 84, in MoveToPosition
    self.basemanip.MoveUnsyncJoints(jointvalues=values,jointinds=indices)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/interfaces/BaseManipulation.py", line 251, in MoveUnsyncJoints
    res = self.prob.SendCommand(cmd)
openravepy._openravepy_0_9.openravepy_ext.openrave_exception: openrave (Failed): CollisionCheckerPQP::DoPQP - ERROR: YOU MUST PASS IN A CollisionReport STRUCT TO MEASURE DISTANCE!

QtDBus: cannot relay signals from parent QObject(0x7fd4b0292950 "") unless they are emitted in the object's thread QThread(0x7fd4b000e4d0 ""). Current thread is QThread(0x21e6330 "").
QtDBus: cannot relay signals from parent QObject(0x7fd4b0292950 "") unless they are emitted in the object's thread QThread(0x7fd4b000e4d0 ""). Current thread is QThread(0x21e6330 "").
[qtcoinrave.cpp:80 DestroyPlugin] SoQt releasing all memory
Coin debug: socontexthandler_cleanup(): 70 context-bound resources not free'd before exit.
QThreadStorage: Thread 0x7fd4b000e4d0 exited after QThreadStorage 6 destroyed
QThreadStorage: Thread 0x7fd4b000e4d0 exited after QThreadStorage 5 destroyed
QThreadStorage: Thread 0x7fd4b000e4d0 exited after QThreadStorage 4 destroyed
QThreadStorage: Thread 0x7fd4b000e4d0 exited after QThreadStorage 3 destroyed
QThreadStorage: Thread 0x7fd4b000e4d0 exited after QThreadStorage 2 destroyed
QThreadStorage: Thread 0x7fd4b000e4d0 exited after QThreadStorage 1 destroyed
Segmentation fault (core dumped)

robotcd, 2015-08-27:
BTW, if I follow the instructions in:
http://openrave.org/docs/latest_stable/coreapihtml/installation_linux.html

sudo add-apt-repository ppa:openrave/release
sudo sh -c 'echo "deb-src http://ppa.launchpad.net/openrave/release/ubuntu `lsb_release -cs` main" >> /etc/apt/sources.list.d/openrave-release-`lsb_release -cs`.list'
sudo apt-get update
sudo apt-get build-dep openrave

I get:
E: Unable to find a source package for openrave
when doing sudo apt-get build-dep openrave

So I found these other instructions:
http://www.aizac.info/installing-openrave0-9-on-ubuntu-trusty-14-04-64bit/

and did:
$sudo apt-get install collada-dom2.4-dp*
$sudo apt-get install libsoqt4-dev
$cd
$git clone --branch latest_stable https://github.com/rdiankov/openrave.git
$cd openrave
$mkdir build
$cd build
$cmake ..
$sudo make install

Is that ok?

Cescuder, 2015-08-28:
Before making the install (cmake & make install), be sure you have installed the packages for all the Openrave plugins, this includes different collision checkers (that's the error your getting).

You'll have to install first ODE(latest version 0.13 works fine for me)  or Bullet (I'm using version 2.80). You have to manually install ODE from his sourceforge repo http://sourceforge.net/projects/opende/files/. You can get Bullet v2.80 adding this ppa to your repo: ppa:joern-schoenyan/libbullet280

robotcd, 2015-08-28:
Thanks @Cescuder ! I actually already had bullet (2.83.6), but didn't have it in my path at the moment of testing. Now I have included it and the hanoi example is running. Awesome!
I will go through the tutorials and examples now, do you recommend any particular tutorial to get started? (python preferred).
Thanks!

robotcd, 2015-08-28:
Please let me know if there is a list of other plugins I should have.

Cescuder, 2015-08-31:
Hi,
The plugins list is just the ccmake plugin list. 
The collision checkers are usually the most conflictive ones, so if you are not having problems with them, I guess you'll also have the other plugins installed/added to the Openrave path.
Regard to a particular tutorial, it depends on what are you interested, I work mainly on manipulation so the basic for me were the grasping module and the simple grasping example.

robotcd, 2015-08-31:
Thanks!
The hanoi example runs fine but I observed these yellow warnings as soon as I launch the example:

[plugindatabase.h:929 _SysLoadLibrary] /usr/local/share/openrave-0.9/plugins/openravepy_configurationcache.so: libconfigurationcache.so: cannot open shared object file: No such file or directory
[plugindatabase.h:857 _LoadPlugin] failed to load: /usr/local/share/openrave-0.9/plugins/openravepy_configurationcache.so
[plugindatabase.h:577 Create] Failed to create name ode, interface collisionchecker
[plugindatabase.h:577 Create] Failed to create name bullet, interface collisionchecker

Not sure if that means I have a problem? 
I thought it was ok because the example run fine but asking just in case.

Cescuder, 2015-10-15:
It seems that some modules(again Ode and Bullet) were not compiled/installed fine, be sure to clean your path and try again to build and install Openrave

Atanker, 2016-07-21:
Hello.I'm a new learner.When I intalled the openrave on ubuntu ,I had a similar problem. When  I run openrve.py --example hanoi,I didn't get the gui with the robot.And it came with the warning :
2016-07-21 14:51:06,432 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
2016-07-21 14:51:06,432 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name ode, interface collisionchecker
robot resolutions:  [ 0.00195998  0.00198425  0.00328496  0.01093304  0.0106936   0.0234112
  0.02297028]
robot weights:  [ 1.52329723  0.92804625  0.38978808  0.15792621  0.10813445  0.04291764
  0.02730175]
hanoimove disk2 from srcpeg to destpeg…………………………
Do you know how to solve it ,please?

undefinedcodezhong, 2020-01-06:
Has anyone solved the problem？？？
I intalled the openrave on ubuntu 16.04, I had a similar problem. When I run openrve.py --example hanoi, it seems to be normally. But when I run openrve.py --example graspplanning, it will throw the error information as follows:
```
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 1
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 5
robot resolutions:  [0.00396017 0.00455076 0.00593835 0.00951787 0.01218716 0.0175349
 0.02203869 0.0606854  0.0606854  0.0606854  0.04497103]
robot weights:  [2.77253268 2.41271402 1.84894655 1.15358696 0.90092279 0.62616198
 0.49820064 0.18092804 0.18092804 0.18092804 0.24415027]
searching for graspable objects (robot=42daa21cdb5bcad0df5d94673fd5d61a)...
attempting to auto-generate a grasp table
2020-01-06 18:51:41,004 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name ode, interface collisionchecker
QtDBus: cannot relay signals from parent QObject(0x7fc99c2b18d0 "") unless they are emitted in the object's thread QThread(0x7fc99c00e4a0 ""). Current thread is QThread(0x2b0a900 "").
QtDBus: cannot relay signals from parent QObject(0x7fc99c2b18d0 "") unless they are emitted in the object's thread QThread(0x7fc99c00e4a0 ""). Current thread is QThread(0x2b0a900 "").
2020-01-06 18:51:41,017 openrave [WARN] [qtcoinrave.cpp:80 DestroyPlugin] SoQt releasing all memory
Coin debug: socontexthandler_cleanup(): 1000 context-bound resources not free'd before exit.
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 149, in <module>
    example.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/openravepy_ext.py", line 64, in newfn
    return fn(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/graspplanning.py", line 476, in run
    OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 240, in parseAndCreateThreadedUser
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 100, in SetViewerUserThread
    userfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 240, in <lambda>
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/graspplanning.py", line 452, in main
    self = GraspPlanning(robot,randomize=options.randomize,nodestinations=options.nodestinations,plannername=options.planner)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/graspplanning.py", line 152, in __init__
    gmodel.autogenerate()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/__init__.py", line 127, in autogenerate
    self.generate(*self.autogenerateparams(options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/grasping.py", line 383, in autogenerateparams
    approachrays = self.computeBoxApproachRays(delta=0.02,normalanglerange=normalanglerange,directiondelta=directiondelta)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/grasping.py", line 969, in computeBoxApproachRays
    return self._computeBoxApproachRays(self.env,self.target,delta=delta,normalanglerange=normalanglerange,directiondelta=directiondelta)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/grasping.py", line 1018, in _computeBoxApproachRays
    cc.DestroyEnvironment()
AttributeError: 'NoneType' object has no attribute 'DestroyEnvironment'
QThreadStorage: Thread 0x7fc99c00e4a0 exited after QThreadStorage 8 destroyed
QThreadStorage: Thread 0x7fc99c00e4a0 exited after QThreadStorage 7 destroyed
QThreadStorage: Thread 0x7fc99c00e4a0 exited after QThreadStorage 4 destroyed
QThreadStorage: Thread 0x7fc99c00e4a0 exited after QThreadStorage 3 destroyed
QThreadStorage: Thread 0x7fc99c00e4a0 exited after QThreadStorage 2 destroyed
QThreadStorage: Thread 0x7fc99c00e4a0 exited after QThreadStorage 1 destroyed
[1]    14568 segmentation fault (core dumped)  openrave.py --example graspplanning
```
Can somebody give me some suggestion to fix it ?