rdiankov/openrave #365: SympifyError with ikfast.py

wenbinli, 2015-09-08:
Hi,

I came across the similar issue to https://github.com/rdiankov/openrave/issues/287
It seems to have been fixed already, but I still got error with sympy 7.4/7.6:

/usr/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py:1122: RuntimeWarning: invalid value encountered in divide
  axisangle /= angle
INFO: moved translation Matrix([[0, 0, 11597/100000]]) to right end
INFO: moved translation Matrix([[0, 0, 5407/20000]]) to left end
INFO: moved translation on intersecting axis Matrix([[0, 0, 2707/10000]]) to left
INFO: [[1, 0, 0, 557/10000],[0, 1, 0, 0],[0, 0, 1, 28139/100000]]
INFO: [[cos(j0), -sin(j0), 0, 0],[sin(j0), cos(j0), 0, 0],[0, 0, 1, 0]]
INFO: [[1, 0, 0, 69/1000],[0, 0, 1, 0],[0, -1, 0, 0]]
INFO: [[cos(j1), -sin(j1), 0, 0],[sin(j1), cos(j1), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, 1, 51/500],[0, 1, 0, 0],[-1, 0, 0, 0]]
INFO: [[cos(j2), -sin(j2), 0, 0],[sin(j2), cos(j2), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, -1, 0],[1, 0, 0, 69/1000],[0, -1, 0, 13121/50000]]
INFO: [[cos(j3), -sin(j3), 0, 0],[sin(j3), cos(j3), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 1, 0, 0],[0, 0, -1, -37429/100000],[-1, 0, 0, 0]]
INFO: [[cos(j4), -sin(j4), 0, 0],[sin(j4), cos(j4), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, -1, 0],[1, 0, 0, 1/100],[0, -1, 0, 0]]
INFO: [[cos(j5), -sin(j5), 0, 0],[sin(j5), cos(j5), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 1, 0, 0],[0, 0, -1, 0],[-1, 0, 0, 0]]
INFO: [[cos(j6), -sin(j6), 0, 0],[sin(j6), cos(j6), 0, 0],[0, 0, 1, 0]]
INFO: [[0, -1, 0, 0],[1, 0, 0, 0],[0, 0, 1, 1747/6250]]
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 6121, in <module>
    chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 1639, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 2100, in solveFullIK_6D
    self.testconsistentvalues = self.computeConsistentValues(jointvars,Tfinal,numsolutions=4)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 1686, in computeConsistentValues
    psubs.append((s,v.subs(psubs)))
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/basic.py", line 892, in subs
    rv = rv._subs(old, new, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/cache.py", line 93, in wrapper
    retval = cfunc(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/compatibility.py", line 872, in wrapper
    result = user_function(*args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/basic.py", line 1006, in _subs
    rv = fallback(self, old, new)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/basic.py", line 978, in fallback
    arg = arg._subs(old, new, **hints)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/cache.py", line 93, in wrapper
    retval = cfunc(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/compatibility.py", line 872, in wrapper
    result = user_function(*args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/basic.py", line 1004, in _subs
    rv = self._eval_subs(old, new)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py", line 300, in Pow_eval_subs
    return Pow(self.base._eval_subs(old, new), self.exp._eval_subs(old, new))
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/cache.py", line 93, in wrapper
    retval = cfunc(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/compatibility.py", line 872, in wrapper
    result = user_function(*args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/power.py", line 162, in __new__
    b = _sympify(b)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/sympify.py", line 350, in _sympify
    return sympify(a, strict=True)
  File "/usr/local/lib/python2.7/dist-packages/sympy/core/sympify.py", line 243, in sympify
    raise SympifyError(a)
sympy.core.sympify.SympifyError: SympifyError: None

rdiankov, 2015-09-09:
unfortunately, ikfast only works with sympy 0.7.1...

wenbinli, 2015-09-09:
Thanks, it works now with sympy 0.7.1.

closed 2015-09-09 by wenbinli