rdiankov/openrave #369: cannot generate ikmodel with default parameters

ompugao, 2015-10-03:
@ 817c72b

````
@ openrave.py -i
openravepy is not set into PYTHONPATH env variable, attempting to add
WARNING: QApplication was not created in the main() thread.
Python 2.7.6 (default, Jun 22 2015, 17:58:13)
Type "copyright", "credits" or "license" for more information.

IPython 1.2.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: env.Load('data/wamtest1.env.xml')
[kinbody.cpp:3232 _ComputeInternalInformation] whiteboard link index 0 has no name
[kinbody.cpp:3232 _ComputeInternalInformation] whiteboard link index 1 has no name
Out[1]: True

In [2]: robot =env.GetRobots()[0]

In [3]: robot.GetManipulators()
Out[3]: [RaveGetEnvironment(1).GetRobot('BarrettWAM').GetManipulator('arm')]

In [4]: manip = robot.GetActiveManipulator()

In [5]: ikmodel = databases.inversekinematics.InverseKinematicsModel(robot)

In [6]: ikmodel.manip
Out[6]: RaveGetEnvironment(1).GetRobot('BarrettWAM').GetManipulator('arm')

In [7]: ikmodel.autogenerate()
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 1
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 5
openravepy.databases.inversekinematics: generate, Generating inverse kinematics for manip arm: Transform6D [0, 1, 3, 4, 5, 6], precision=None, maxcasedepth=3 (this might take up to 10 min)
openravepy.databases.inversekinematics: generate, creating ik file /home/sifi/.openrave/kinematics.e9a051e4825529aa31892beb41684ca4/ikfast0x10000048.Transform6D.0_1_3_4_5_6_f2.cpp
/home/sifi/install/lib/python2.7/site-packages/openravepy/_openravepy_0_9/ikfast.py:1437: RuntimeWarning: invalid value encountered in divide
  axisangle /= angle
openravepy.ikfast: forwardKinematicsChain, moved translation Matrix([[0, 0, 0]]) to right end
openravepy.ikfast: forwardKinematicsChain, moved translation Matrix([[0, 0, 0]]) to left end
openravepy.ikfast: forwardKinematicsChain, moved translation on intersecting axis Matrix([[0, 0, 3/10]]) to left
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 11/50],[0, 1, 0, 7/50],[0, 0, 1, 173/500]]
openravepy.ikfast: generateIkSolver, [[cos(j0), -sin(j0), 0, 0],[sin(j0), cos(j0), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, 1, 0],[0, -1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j1), -sin(j1), 0, 0],[sin(j1), cos(j1), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, -1, 0],[0, 1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j2), -sin(j2), 0, 0],[sin(j2), cos(j2), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 9/200],[0, 0, 1, 0],[0, -1, 0, 11/20]]
openravepy.ikfast: generateIkSolver, [[cos(j3), -sin(j3), 0, 0],[sin(j3), cos(j3), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, -9/200],[0, 0, -1, -3/10],[0, 1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j4), -sin(j4), 0, 0],[sin(j4), cos(j4), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, 1, 0],[0, -1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j5), -sin(j5), 0, 0],[sin(j5), cos(j5), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, -1, 0],[0, 1, 0, 0]]
openravepy.ikfast: generateIkSolver, [[cos(j6), -sin(j6), 0, 0],[sin(j6), cos(j6), 0, 0],[0, 0, 1, 0]]
openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, 3/50]]
openravepy.ikfast: solveFullIK_6D, ikfast 6d: [j0, j1, j3, j4, j5, j6]
openravepy.ikfast: iterateThreeIntersectingAxes, found 3 consecutive intersecting axes links[8:13], rotvars=[j4, j5, j6], translationvars=[j0, j1, j3]
openravepy.ikfast: SolveAllEquations, depth=0 c=1, [j2] [j0, j1, j3]: cases=None
openravepy.ikfast: SolveAllEquations, depth=0 c=3, [j2, j3] [j0, j1]: cases=None
openravepy.ikfast: checkForDivideByZero, adding atan2(-py, px) = Abs(px) + Abs(py) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(-py, px) = Abs(px) + Abs(py) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(-3*cj3/10 - 9*sj3/200 - 11/20, 3*cj2*(-3*cj3 + 20*sj3 + 3)/200) = 3*Abs(cj2*(-3*cj3 + 20*sj3 + 3))/200 + Abs(-3*cj3/10 - 9*sj3/200 - 11/20) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(-3*cj3/10 - 9*sj3/200 - 11/20, 3*cj2*(-3*cj3 + 20*sj3 + 3)/200) = 3*Abs(cj2*(-3*cj3 + 20*sj3 + 3))/200 + Abs(-3*cj3/10 - 9*sj3/200 - 11/20) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(py**2*(cj2**2 + sj2**2), -px*py*(cj2**2 + sj2**2)) = Abs(py**2*(cj2**2 + sj2**2)) + Abs(px*py*(cj2**2 + sj2**2)) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(py**2*(cj2**2 + sj2**2), -px*py*(cj2**2 + sj2**2)) = Abs(py**2*(cj2**2 + sj2**2)) + Abs(px*py*(cj2**2 + sj2**2)) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(px*py*(cj2**2 + sj2**2), -px**2*(cj2**2 + sj2**2)) = Abs(px**2*(cj2**2 + sj2**2)) + Abs(px*py*(cj2**2 + sj2**2)) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(px*py*(cj2**2 + sj2**2), -px**2*(cj2**2 + sj2**2)) = Abs(px**2*(cj2**2 + sj2**2)) + Abs(px*py*(cj2**2 + sj2**2)) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(-py, px) = Abs(px) + Abs(py) all zeros check
[... 753 lines cut ...]
openravepy.ikfast: solveSingleVariable, j4 solution: equations used for atan2: [new_r12*cos(j4), new_r12*sin(j4), gconst14*sin(j4), -new_r00*sin(j4), gconst14*sin(j4) + new_r01, gconst14 + new_r00*cos(j4), gconst14*cos(j4) + new_r00, -gconst14*cos(j4) + new_r11, new_r01*cos(j4) + new_r11*sin(j4), -gconst14 - new_r01*sin(j4) + new_r11*cos(j4)]
openravepy.ikfast: checkForDivideByZero, adding atan2(-new_r01, new_r11) = Abs(new_r01) + Abs(new_r11) all zeros check
openravepy.ikfast: solveSingleVariable, j4 solution: equations used for atan2: [new_r12*cos(j4), new_r12*sin(j4), gconst14*sin(j4), -new_r00*sin(j4), gconst14*sin(j4) + new_r01, gconst14 + new_r00*cos(j4), gconst14*cos(j4) + new_r00, -gconst14*cos(j4) + new_r11, new_r01*cos(j4) + new_r11*sin(j4), -gconst14 - new_r01*sin(j4) + new_r11*cos(j4)]
openravepy.ikfast: checkForDivideByZero, adding atan2(-new_r01, new_r11) = Abs(new_r01) + Abs(new_r11) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(-new_r01, -new_r00) = Abs(new_r00) + Abs(new_r01) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(-new_r01, new_r11) = Abs(new_r01) + Abs(new_r11) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(-new_r01, new_r11) = Abs(new_r01) + Abs(new_r11) all zeros check
openravepy.ikfast: AddSolution, c=192, 3 levels deep in checking degenerate cases, skipping. curvars=[j4], AllEquations=[new_r22 + 1, new_r22 + 1, new_r20, new_r20, 0, new_r12, new_r21, new_r21, new_r12*cos(j4), new_r12*cos(j4), new_r12*sin(j4), -new_r00*sin(j4), -gconst14 - new_r01*sin(j4) + new_r11*cos(j4), -new_r00*sin(j4), -gconst14 - new_r01*sin(j4) + new_r11*cos(j4), new_r01*cos(j4) + new_r11*sin(j4), gconst14*sin(j4) + new_r01, gconst14 + new_r00*cos(j4), gconst14*cos(j4) + new_r00, -gconst14*cos(j4) + new_r11, gconst14*sin(j4), -new_r12*sin(j4), new_r20, new_r21, -new_r01*cos(j4) - new_r11*sin(j4), new_r22 + 1, -gconst14 - new_r00*cos(j4)]
openravepy.ikfast: AddSolution, depth=2, c=190, iter=0/6, adding newcases: set([Abs(j6 + atan2(new_r02, new_r12)), Abs(j5 - 3.14159265358979323846264338328), Abs(new_r02) + Abs(new_r10)])
openravepy.ikfast: AddSolution, depth=2, c=190, iter=1/6, starting newcases: set([Abs(new_r02) + Abs(new_r12), Abs(j6 + atan2(new_r02, new_r12)), Abs(j5 - 3.14159265358979323846264338328)])
openravepy.ikfast: AddSolution, depth=2, c=190, iter=2/6, starting newcases: set([Abs(new_r00) + Abs(new_r10), Abs(j6 + atan2(new_r02, new_r12)), Abs(j5 - 3.14159265358979323846264338328)])
openravepy.ikfast: SolveAllEquations, depth=3 c=193, [j0, j1, j3, j2, j5, j6] [j4]: cases=set([Abs(new_r00) + Abs(new_r10), Abs(j6 + atan2(new_r02, new_r12)), Abs(j5 - 3.14159265358979323846264338328)])
openravepy.ikfast: solveSingleVariable, j4 solution: equations used for atan2: [gconst13*sin(j4) + gconst14*cos(j4), -new_r01*sin(j4) + new_r11*cos(j4), -gconst13*cos(j4) + gconst14*sin(j4), -new_r01*cos(j4) - new_r11*sin(j4), -gconst13*cos(j4) + gconst14*sin(j4) + new_r01, -gconst13 + new_r01*cos(j4) + new_r11*sin(j4), -gconst13*sin(j4) - gconst14*cos(j4) + new_r11,  [... cut ...]
openravepy.ikfast: checkForDivideByZero, adding atan2(-new_r11**2, -new_r01*new_r11) = Abs(new_r11**2) + Abs(new_r01*new_r11) all zeros check
openravepy.ikfast: checkForDivideByZero, adding atan2(new_r11**2, new_r01*new_r11) = Abs(new_r11**2) + Abs(new_r01*new_r11) all zeros check
openravepy.ikfast: AddSolution, c=194, 3 levels deep in checking degenerate cases, skipping. curvars=[j4], AllEquations=[1, 1, new_r20, new_r20, new_r11, -new_r01, 0, 0, -new_r01*cos(j4) - new_r11*sin(j4), -new_r01*cos(j4) - new_r11*sin(j4), -new_r01*sin(j4) + new_r11*cos(j4), -gconst13, -gconst14 - new_r01*sin(

rdiankov, 2015-10-04:
that's weird, i just tested it on latest production and it works
can you check your sympy version?

ompugao, 2015-10-04:
ah, I've totally forgotten openrave requires a specific version of sympy

sudo apt-get remove python-sympy
sudo pip install sympy==0.7.1
fixed this problem
i'm sorry to bother you

closed 2015-10-04 by ompugao