rdiankov/openrave #373: ikfast.py says TypeError

Puttichai, 2015-10-30:
When I tried to generate an IK model for a robot by

```python
iktype = IkParameterization.Type.Transform6D
ikmodel = databases.inversekinematics.InverseKinematicsModel(robot, iktype=iktype)
if not ikmodel.load():
    ikmodel.autogenerate()
```
I have got this error:

```python
/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_/ikfast.py in forwardKinematicsChain(self, chainlinks, chainjoints)
   1613                         angle = sqrt(axisangle[0]**2+axisangle[1]**2+axisangle[2]**2)
   1614                         if angle > 0:
-> 1615                             axisangle /= angle
   1616                         log.debug('rotation angle of Links[%d]: %f, axis=[%f,%f,%f]', len(Links), (angle*180/pi).evalf(),axisangle[0],axisangle[1],axisangle[2])
   1617                     Links.append(self.RoundMatrix(Tright * TLeftjoint))

TypeError: ufunc 'divide' output (typecode 'O') could not be coerced to provided output parameter (typecode 'd') according to the casting rule ''same_kind''
```

I think this is because this is not compatible with a newer version of numpy. (I have Numpy 1.10.1 installed on my computer.)

Do you have any suggestion to this?
Thank you.

rdiankov, 2015-11-12:
that's weird, what's the actual value of angle? can you print?
thanks

sudarshan85, 2016-02-16:
I'm also having the same problem. I'm working with Numpy version 1.10.4 and I built and installed Openrave from source on Ubuntu 14.04.

rdiankov, 2016-02-18:
guys, this problem is fixed in the latest master as of 2016/02/18

thanks for the report!

closed 2016-02-18 by rdiankov