rdiankov/openrave #376: ikfast regression: "symbolic boolean expression has no truth value."

benaxelrod, 2015-11-17:
I had previously been using a fork of master branch from June 12, 2015 to to generate Transform6D ikfast files for my 8 DOF robot quite successfully.  Today (November 17, 2015) I pulled the master branch and regenerated my ikfast files.  However, I now get this error:

```
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 126, in <module>
    database.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1126, in run
    InverseKinematicsModel.RunFromParser(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1091, in RunFromParser
    model = DatabaseGenerator.RunFromParser(Model=Model,parser=parser,robotatts=robotatts,args=args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/__init__.py", line 262, in RunFromParser
    model.autogenerate(options=options)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 661, in autogenerate
    self.generate(iktype=iktype,freejoints=freejoints,precision=precision,forceikbuild=forceikbuild,outputlang=outputlang,ipython=ipython,ikfastmaxcasedepth=ikfastmaxcasedepth)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 887, in generate
    chaintree = solver.generateIkSolver(baselink=baselink,eelink=eelink,freeindices=self.freeindices,solvefn=solvefn)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 2264, in generateIkSolver
    chaintree = solvefn(self, LinksRaw, jointvars, isolvejointvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 754, in solveFullIK_6D
    return self.ikfast.IKFastSolver.solveFullIK_6D(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 2858, in solveFullIK_6D
    tree = self.TestIntersectingAxes(solvejointvars,Links, LinksInv,endbranchtree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 2946, in TestIntersectingAxes
    return self.solve6DIntersectingAxes(T0links,T1links,transvars,rotvars,solveRotationFirst=solveRotationFirst, endbranchtree=endbranchtree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 3141, in solve6DIntersectingAxes
    rottree += self.SolveAllEquations(AllEquations,curvars=currotvars,othersolvedvars=othersolvedvars,solsubs=self.freevarsubs[:],endbranchtree=storesolutiontree)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 6622, in SolveAllEquations
    return self.AddSolution(solutions,AllEquations,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 6721, in AddSolution
    return [solution[0].subs(solsubs)]+self.SolveAllEquations(AllEquations,curvars=newvars,othersolvedvars=othersolvedvars+[var],solsubs=solsubs+self.Variable(var).subs,endbranchtree=endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 6511, in SolveAllEquations
    return self.AddSolution(solutions,AllEquations,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 7283, in AddSolution
    newtree = self.SolveAllEquations(NewEquationsClean,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=newcases, currentcasesubs=newcasesubs, unknownvars=unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 6622, in SolveAllEquations
    return self.AddSolution(solutions,AllEquations,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=currentcases, currentcasesubs=currentcasesubs, unknownvars=unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 7283, in AddSolution
    newtree = self.SolveAllEquations(NewEquationsClean,curvars,othersolvedvars,solsubs,endbranchtree,currentcases=newcases, currentcasesubs=newcasesubs, unknownvars=unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 6602, in SolveAllEquations
    rawsolutions=self.SolvePairVariables(raweqns,var0,var1,othersolvedvars,unknownvars=curvars+unknownvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 8709, in SolvePairVariables
    return self.SolvePairVariablesHalfAngle(raweqns,var0,var1,othersolvedvars)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 7512, in SolvePairVariablesHalfAngle
    possiblefinaleq = self.checkFinalEquation(Poly(Malldet,leftvar),subs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py", line 7915, in checkFinalEquation
    if len(coeffs) == 1 and Abs(coeffs[0]) < 2*(10.0**-self.precision):
  File "/usr/lib/python2.7/dist-packages/sympy/core/relational.py", line 226, in __nonzero__
    raise TypeError("symbolic boolean expression has no truth value.")
TypeError: symbolic boolean expression has no truth value.
```

(Sorry, line numbers might be off due to added print statements).  I printed out ```coeffs``` and most of the time it is just a number, but it was ```[1.0*new_r00]``` when it threw the error.

rdiankov, 2015-11-17:
i actually made several commits yesterday that could have broken it. 
can you send me the robot file?
thanks

benaxelrod, 2015-11-25:
I sent Rosen my robot file and this appears to have been fixed in commit [ba984da](https://github.com/rdiankov/openrave/commit/ba984da9946b2585be75d6caf8fe7a4ad2b771d7). However, it takes *a lot* longer to generate the ikfast file.

closed 2015-11-25 by benaxelrod

matt12345678, 2016-01-12:
I'm having the same issue when I run openrave.py --database inversekinematics --robot=/home/ur5/openrave/src/robots/pr2-beta-static.zae --manipname=rightarm --freejoint=r_shoulder_pan_joint --freeinc=0.01

I'm using the fix from ba984da. Any advice? Thank you.