rdiankov/openrave #378: Can't generate transform6d for Kinova's Jaco

mlauret, 2015-11-20:
Hello !
I am stuck on a problem for quite some time : 
I try to generate a ikfast .cpp for Kinova's Jaco V2 with 6 DOF.
I'm using the latest version of Openrave (branch Master) 0.9 on a Ubuntu 14.04 32Bits with 4GB of RAM.
I didn't succeed with the 2 attempt described below, I think it's a problem with the Jaco2 special kinematics but i'm not sure. Maybe you can understand if it's a bug from openrave or from me.

The .dae file i'm using is the one I created from the Kinova's Github : https://github.com/Kinovarobotics/kinova-ros/blob/master/jaco_model/urdf/standalone_arm.xacro

I used urdf_to_collada to generate the .dae, then I rounded it to 3 digits : 
https://drive.google.com/file/d/0B0BRMr7Z9cxfR1FDcDdRbjJua1U

Everything seem fine with ```openrave-robot.py jaco_arm.rounded3.dae --info links``` : 

name    |           index   |   parents       
------------|-----------------|----------
root        |       0     |              
jaco_link_base  |   1  |   root          
jaco_link_1     |   2    | jaco_link_base
jaco_link_2     |   3   |  jaco_link_1   
jaco_link_3    |    4   |  jaco_link_2   
jaco_link_4     |   5  |   jaco_link_3   
jaco_link_5     |   6 |    jaco_link_4   
jaco_link_hand  |   7   |  jaco_link_5   
jaco_link_finger_1 | 8 |    jaco_link_hand
jaco_link_finger_2 |  9  |   jaco_link_hand
jaco_link_finger_3 |  10  |  jaco_link_hand

Then I launch : 
```
~/catkin_ws/src/kinova-ros$ python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot=jaco_arm.rounded3.dae --iktype=transform6d --baselink=1 --eelink=7 --savefile=jaco_arm.cpp
```

The generation don't work, it compute for ~40min and when openrave ate all my RAM, it crash.
The full log can be found here : http://justpaste.it/jacoikfastcrash (I don't know what's important or not, so I pasted everything... brave yourself, big equation are coming).

The log begin by : 
```
[colladareader.cpp:2184 FillGeometryColor] transparecy is 1.000000, which means the item will be rendered invisible, this must be a mistake so setting to opaque (1)
INFO: moved translation [0, 0, 139247*1111110149**(1/2)/166666522350 + 717062502*4139417560063315159718369105**(1/2)/1218527987716250020523125] to right end
INFO: moved translation [0, 0, 59/500] to left end
INFO: moved translation on intersecting axis [0, 0, 33/500] to left
INFO: [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, 271/1000]]
INFO: [[cos(j0), -sin(j0), 0, 0],[sin(j0), cos(j0), 0, 0],[0, 0, 1, 0]]
INFO: [[-1, 0, 0, 0],[0, 0, 1, 0],[0, 1, 0, 0]]
INFO: [[cos(j1), -sin(j1), 0, 0],[sin(j1), cos(j1), 0, 0],[0, 0, 1, 0]]
INFO: [[-1, 0, 0, -41/100],[0, -1, 0, 0],[0, 0, 1, 0]]
INFO: [[cos(j2), -sin(j2), 0, 0],[sin(j2), cos(j2), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 1, 0, 0],[0, 0, 1, 273/1000],[1, 0, 0, 11/1000]]
INFO: [[cos(j3), -sin(j3), 0, 0],[sin(j3), cos(j3), 0, 0],[0, 0, 1, 0]]
INFO: [[-29160370163987151/50839629836012849, 0, 41645417200000000/50839629836012849, 17/500],[0, -1, 0, 0],[41645417200000000/50839629836012849, 0, 29160370163987151/50839629836012849, 0]]
INFO: [[cos(j4), -sin(j4), 0, 0],[sin(j4), cos(j4), 0, 0],[0, 0, 1, 0]]
INFO: [[57365*1111110149**(1/2)/3333330447, 4*1111110149**(1/2)/3333330447, -81910*1111110149**(1/2)/3333330447, -17/500 - 81910*1111110149**(1/2)*(-717062502*4139417560063315159718369105**(1/2)/1218527987716250020523125 - 139247*1111110149**(1/2)/166666522350)/3333330447],[-6730909694*3086414414198440541983105405**(1/2)/9259243242595321625949316215, 33333304469104*3086414414198440541983105405**(1/2)/1851848648519064325189863243, 3425069647*3086414414198440541983105405**(1/2)/9259243242595321625949316215, 3425069647*3086414414198440541983105405**(1/2)*(-717062502*4139417560063315159718369105**(1/2)/1218527987716250020523125 - 139247*1111110149**(1/2)/166666522350)/9259243242595321625949316215],[1365166668*4139417560063315159718369105**(1/2)/107230462919030001806035, 7097*4139417560063315159718369105**(1/2)/21446092583806000361207, 956083336*4139417560063315159718369105**(1/2)/107230462919030001806035, 956083336*4139417560063315159718369105**(1/2)*(-717062502*4139417560063315159718369105**(1/2)/1218527987716250020523125 - 139247*1111110149**(1/2)/166666522350)/107230462919030001806035 + 33/500]]
INFO: [[cos(j5), -sin(j5), 0, 0],[sin(j5), cos(j5), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 1, 0, 0],[1, 0, 0, 0],[0, 0, -1, 139247*1111110149**(1/2)/166666522350 + 717062502*4139417560063315159718369105**(1/2)/1218527987716250020523125]]
INFO: ikfast 6d: [j0, j1, j2, j3, j4, j5]
INFO: found 3 consecutive non-intersecting axes links[2:7], vars=[j1, j2, j3]
INFO: found 3 consecutive non-intersecting axes links[4:9], vars=[j2, j3, j4]
INFO: found 3 consecutive non-intersecting axes links[6:11], vars=[j3, j4, j5]
INFO: found 3 consecutive non-intersecting axes links[0:5], vars=[j0, j1, j2]
INFO: try first group 0/4
INFO: computed in 26.591491s
WARNING: raghavan roth equation (11,1) too complex
WARNING: raghavan roth equation (12,1) too complex
WARNING: raghavan roth equation (13,1) too complex
INFO: computed in 245.490833s
```
And end by : 
```
INFO: attempting li/woernle/hiller general ik method
INFO: allowed indices: [0]
[... 180 lines cut ...]
INFO: not unique: (Poly((-801*3**(1/2)/25000 + 37/2000)*cj3*sj4*htj2 + (-213600099591*3**(1/2)/13333330225000 + 9866671267/1066666418000)*sj3*cj4*htj2 + (-8544780*3**(1/2)/533333209 + 41931281533/266666604500)*sj3*htj2 + (-36996588*3**(1/2)/2666666045 + 4272390/533333209)*cj4*htj2**2 + (-4272390/533333209 + 36996588*3**(1/2)/2666666045)*cj4 + (-7869598613559/26666660450000 + 9866671267*3**(1/2)/2133332836000)*htj2**2, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(-px*cj0*sj1*htj2**2 - px*cj0*sj1 - py*sj0*sj1*htj2**2 - py*sj0*sj1 - pz*cj1*htj2**2 - pz*cj1 - 7869598613559/26666660450000 + 9866671267*3**(1/2)/2133332836000, cj0, sj0, cj1, sj1, htj2, domain='EX'))
INFO: assuming equation htj2 is non-zero
INFO: not unique: [Poly(266666791*3**(1/2)/1066666418*cj3*cj4 + 230940000/533333209*cj3 - 3**(1/2)/2*sj3*sj4, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(r12*cj0 - r02*sj0, cj0, sj0, cj1, sj1, htj2, domain='ZZ[r02,r12]')]
INFO: assuming equation htj2 is non-zero
INFO: not unique: (Poly(266666791*3**(1/2)/1066666418*cj3*cj4 + 230940000/533333209*cj3 - 3**(1/2)/2*sj3*sj4, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(r12*cj0 - r02*sj0, cj0, sj0, cj1, sj1, htj2, domain='ZZ[r02,r12]'))
INFO: not unique: [Poly((-213600099591*3**(1/2)/26666660450000 + 9866671267/2133332836000)*cj3*cj4 + (-4272390*3**(1/2)/533333209 + 41931281533/533333209000)*cj3 + (-37/4000 + 801*3**(1/2)/50000)*sj3*sj4, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(py*cj0 - px*sj0 - 9/800, cj0, sj0, cj1, sj1, htj2, domain='QQ[px,py]')]
INFO: assuming equation htj2 is non-zero
INFO: not unique: (Poly((-213600099591*3**(1/2)/26666660450000 + 9866671267/2133332836000)*cj3*cj4 + (-4272390*3**(1/2)/533333209 + 41931281533/533333209000)*cj3 + (-37/4000 + 801*3**(1/2)/50000)*sj3*sj4, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(py*cj0 - px*sj0 - 9/800, cj0, sj0, cj1, sj1, htj2, domain='QQ[px,py]'))
INFO: assuming equation htj2 is non-zero
INFO: not unique: (Poly((-9866671267/1066666418000 + 161663974263*3**(1/2)/1066666418000)*cj3*sj4 + (-37/2000 + 123553585737*3**(1/2)/1066666418000)*sj3*cj4 + 115339759133/1066666418000*sj3, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(-41*r02/100*cj0*sj1 + rxp2_1*cj0 - 41*r12/100*sj0*sj1 - rxp2_0*sj0 - 41*r22/100*cj1, cj0, sj0, cj1, sj1, htj2, domain='QQ[r02,r12,r22,rxp2_0,rxp2_1]'))
INFO: assuming equation htj2 is non-zero
INFO: not unique: (Poly((-1922400896319*3**(1/2)/10666664180000000 + 88800041403/853333134400000)*cj3*cj4 + (-3845151*3**(1/2)/21333328360 + 377381533797/213333283600000)*cj3 + (-333/1600000 + 7209*3**(1/2)/20000000)*sj3*sj4 + (-4267571087921/1066666418000000 + 92387147065533*3**(1/2)/13333330225000000)*cj4, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(-41*px/50*cj0*cj1 - 41*py/50*sj0*cj1 + 41*pz/50*sj1 + pp + 4571482672269*3**(1/2)/1066666418000000 + 392480205673429931/5333332090000000000, cj0, sj0, cj1, sj1, htj2, domain='EX'))
INFO: assuming equation htj2 is non-zero
INFO: not unique: (Poly(2400001119*3**(1/2)/853333134400*cj3*cj4 + 2598075/533333209*cj3 - 9*3**(1/2)/1600*sj3*sj4 - 115339759133*3**(1/2)/1066666418000*cj4, cj3, sj3, cj4, sj4, htj2, domain='EX'), Poly(-41*r02/100*cj0*cj1 - 41*r12/100*sj0*cj1 + 41*r22/100*sj1 + npz - 2234439842607/26666660450000, cj0, sj0, cj1, sj1, htj2, domain='QQ[r02,r12,r22,npz]'))
INFO: assuming equation htj2 is non-zero
[... cut ...] p*r12 + 1681*r12/10000)*cj0 + 41*rxp2_1/50*sj0*sj1 + (2*npz*px - pp*r02 - 1681*r02/10000)*sj0 + 41*rxp2_2/50*cj1 + 20109958583463/10666664180000000, cj0, sj0, cj1, sj1, htj2, domain='QQ[r02,r12,npz,pp,px,py,rxp2_0,rxp2_1,rxp2_2]'))
INFO: matrix has 0 symbols
INFO: skipping dependent index 11
INFO: skipping dependent index 15

```

And now it had been running for 1h and a half and taking only 14% of my RAM. Nothing more appeared. Do all these log mean something to you ?

Thanks you for your time!

RYANZYC, 2019-04-18:
HI! I have a similar problem WARNING: equation too complex to simplify for rot norm.not unique.Have you solved it?

wustdemg, 2019-05-03:
I also wander if it is my openrave's problem?

anubhav1772, 2019-11-16:
Hi, I am getting similar errors/warnings. Have you able to solve this ?

```
WARNING: equation too complex to simplify for rot norm
WARNING: raghavan roth equation (11,1) too complex
WARNING: raghavan roth equation (12,1) too complex
WARNING: raghavan roth equation (13,1) too complex
```
I would really appreciate some help.