rdiankov/openrave #379: Recording videos

stephane-caron, 2015-11-25:
Hi there,

The video recording feature (e.g. from the QtCoin GUI: "Options → Record Real-time Video", or using the [code snippet from the examples](http://openrave.org/docs/latest_stable/tutorials/openravepy_examples/#recording-videos)) does not seem to work on a fresh install on Ubuntu 14.04 ([installation details here](https://scaron.info/teaching/installing-openrave-on-ubuntu-14.04.html), confirmed today with the latest commit from ``master``). Whenever started, it outputs:

```
[qtcoinviewer.cpp:3514 _GetVideoFrame] offset buffer null, disabling
[qtcoinviewer.cpp:2887 _VideoFrame] cannot record video
[qtcoinviewer.cpp:2887 _VideoFrame] cannot record video
[qtcoinviewer.cpp:2887 _VideoFrame] cannot record video
[qtcoinviewer.cpp:2887 _VideoFrame] cannot record video
(... goes on and on ...)
```

Any idea on how to debug this?

joeyzhouyue, 2016-04-09:
Hello, did you fixed the problem?
I have the save problem as yours.

Dadzemovic, 2017-04-21:
Also experiencing this issue.

stephane-caron, 2019-04-06:
I ended up implementing another way of making videos via a [CameraRecorder](https://github.com/stephane-caron/pymanoid/blob/c0f17db6d8b7f3028b9d4b72e16c2691e93ef295/pymanoid/sim.py#L537) class that takes screenshots of every simulation step and post-processes them into an mpeg4 video using [avconv](https://manpages.ubuntu.com/manpages/precise/man1/avconv.1.html). It makes the video recording slow of course, but provides a smooth video in the end; see for instance the [OpenRAVE clips inserted in this video](https://scaron.info/videos/capture-walking.mp4).

Feel free to adapt it to your needs!