rdiankov/openrave #394: suggestion: create set official docker images

gavanderhoorn, 2016-04-21:
Seeing as the installation of OpenRAVE on recent Linux distributions is complex (at least to get right, with specific versions of dependencies required, etc), it might be nice to provide a set of Docker images (through the hub). I've searched a bit and there are a few out there, but they are either not maintained, or are user contributed (which is fine).

An 'official' set of images would make setup of OpenRAVE a breeze, even if they are just base images with no 3D acceleration (which users would have to layer on themselves).

Neon22, 2016-04-22:
The inkscape people were recently playing with something called Appimage which was working well for them...

ziyan, 2016-09-05:
The problem with an official OpenRAVE docker image would be having to include adequate built-in graphics driver support, which will make the image very large.

gavanderhoorn, 2016-09-05:
@ziyan: not necessarily. A base image that includes a pre-built version of OpenRAVE, but not any hw-specific graphics drivers would already be a huge timesaver.

probonopd, 2016-09-05:
Have a look at the [Krita AppImage](https://krita.org/en/download/krita-desktop/); I think it uses the native OS graphics drivers intensively and seems to be working well.

gavanderhoorn, 2016-09-05:
@probonopd: seems you are the author of AppImage?

probonopd, 2016-09-05:
@gavanderhoorn yes, direct to me any questions or doubts you might have regarding it.

gavanderhoorn, 2016-09-05:
I don't have any. I just thought I'd mention you are the author. Your comment made it look like you were "just another user".