rdiankov/openrave #397: Failed to create name fcl_, interface collisionchecker

conversAlgoDS, 2016-05-10:
I managed to compil openrave  on my fedora machine but when I run the hanoi example., it fails to display the gui. There is only 1 warning about fcl. Messages to console are however getting displayed.

```
$openrave.py --example hanoi
[plugindatabase.h:577 Create] Failed to create name fcl_, interface collisionchecker
robot resolutions:  [ 0.00195998  0.00198425  0.00328496  0.01093304  0.0106936   0.0234112
  0.02297028]
robot weights:  [ 1.52329723  0.92804625  0.38978808  0.15792621  0.10813445  0.04291764
  0.02730175]
hanoimove disk2 from srcpeg to destpeg
opening hand
moving hand to location
succeeded so grab the disk
try to put the disk in the destination peg
move to position above source peg
move to position above dest peg
move to dest peg
wait for robot to complete all trajectories
done with one disk
hanoimove disk1 from srcpeg to peg
opening hand
moving hand to location
succeeded so grab the disk
try to put the disk in the destination peg
.
.
.
```

hustfzt2, 2016-05-28:
Have you solved for this problem？

rdiankov, 2016-05-28:
sounds like you failed to compile the openscenegraph plugin or the soqt plugin for the viewers....

Atanker, 2016-07-21:
Hello,have you solved it?I have the same problem,and more ,it shows "failed to creat name ode,interface collisionchecker"

Cescuder, 2016-09-05:
Hi @Atanker 
It sounds that the ode plugin is missing, have you installed it before compiling?

You can download and install it manually, then compile openrave.

yijiangh, 2017-01-29:
In my scenario, just making sure you have install the correct version of FCL will fix this problem.
Check FCL installation instruction from [Stéphane Caron](https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html)