rdiankov/openrave #399: Installation problem

mgvargas, 2016-05-12:
I am new to openrave, I'm having some problems while installing it. Using 
 git clone --branch latest_stable https://github.com/rdiankov/openrave.git

I get the following mistake while building it:
`make[2]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/fclrave.cpp.o] Error 1`
`make[1]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/all] Error 2`
`make: *** [all] Error 2`

The whole error is quite long, it starts with:
`/home/minerva/Programs/openrave/plugins/fclrave/fclcollision.h:204:17: warning: identifier ‘nullptr’ is a keyword in C++11 [-Wc++0x-compat]`
`                 std::ostream nullout(nullptr);`
`                 ^`
`In file included from /home/minerva/Programs/openrave/plugins/fclrave/fclcollision.h:8:0,`
                ` from /home/minerva/Programs/openrave/plugins/fclrave/fclrave.cpp:18:`
`/home/minerva/Programs/openrave/plugins/fclrave/fclspace.h:11:9: error: ‘shared_ptr’ does not name a type`
 `typedef shared_ptr<fcl::CollisionGeometry> CollisionGeometryPtr`

If I comment that plugin in the plugins/CMakeList.txt
 #

rdiankov, 2016-05-15:
what simulations don't work for you?

we're using latest versions of libraries for FCL and OpenSceneGraph. if you don't want to play around with it, try building with (or as you did, comment out the lines in the CMakeLists.txt)

`
-DOPT_FCL_COLLISION=OFF -DOPT_QTOSG_VIEWER=OFF
`

rdiankov, 2016-05-15:
FYI, i just pushed some stuff to master, can you try it out?

mgvargas, 2016-05-16:
Thanks a lot for your reply!

The problem with fclrave keeps appearing, also the one with qtosgrave
`make[2]: *** [plugins/qtosgrave/CMakeFiles/qtosgrave.dir/qtosgviewer.cpp.o] Error 1`
`make[1]: *** [plugins/qtosgrave/CMakeFiles/qtosgrave.dir/all] Error 2`
`make: *** [all] Error 2`

If I comment the plugin: 
`#      set(PLUGINS ${PLUGINS} qtosgrave)`
the error with FCL appears again:
`make[2]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/fclrave.cpp.o] Error 1`
`make[1]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/all] Error 2`
`make: *** [all] Error 2`

I tried installing FCL from https://github.com/flexible-collision-library/fcl.git

The simulations that are not working are hanoi and graspplanning:
`2016-05-16 18:28:16,697 openrave [ERROR] [ikfastsolver.cpp:215 Init] inverse kinematics hashes do not match for manip Puma:arm. IK will not work! 74458059fde60df7401cc1dfcfa06d15!=a6c70e6dd694838553470dde754d5825`
[https://drive.google.com/open?id=0B9gd9T5QCnR_QUtRTUo4V3pZMkU](url)

Any ideas of what am I doing wrong?

Thanks

flyyflyy, 2016-05-23:
I am also new to openrave and I'm having the same problems with qtosgrave and flcrave. 

Did you thind a way to avoid these problems ?

Thanks

mgvargas, 2016-05-23:
I installed an older version of openrave.

The version I used was d3de4352cc3e61292938dbd8e2c1d372e814710b, this one has no problems with flcrave, so I only commented the qtosgrave plugin.

I hope that works for you too.