rdiankov/openrave #400: make error when building qtosgrave plugin with Ubuntu 14.04.4

WHAAAT, 2016-05-15:
Hello,
normally i had no issue building openrave from source. 
However since the newest Ubuntu I'm getting the following error.
```

/openrave/plugins/qtosgrave/qtosgviewer.cpp: In member function ‘virtual void qtosgrave::QtOSGViewer::_SetTriangleMesh(const float*, int, const int*, int, osg::ref_ptr<osg::TemplateArray<osg::Vec3f, (osg::Array::Type)28u, 3, 5126> >, osg::ref_ptr<osg::DrawElementsUInt>)’:
/home/megatron/openrave/plugins/qtosgrave/qtosgviewer.cpp:1554:17: error: ‘class osg::DrawElementsUInt’ has no member named ‘resizeElements’
     osgindices->resizeElements(numTriangles*3);
                 ^
make[2]: *** [plugins/qtosgrave/CMakeFiles/qtosgrave.dir/qtosgviewer.cpp.o] Error 1
make[1]: *** [plugins/qtosgrave/CMakeFiles/qtosgrave.dir/all] Error 2
make: *** [all] Error 2
```
I've tried different older revisions with same error.

When I disable the plugin I'm getting errors when building plugins/fclrave/CMakeFiles/fclrave.dir/fclrave.cpp.o
this seems to be the same problem like [issues/399](https://github.com/rdiankov/openrave/issues/399)

[here is my full log of the error](https://github.com/rdiankov/openrave/files/264786/log.log.zip)

rdiankov, 2016-05-15:
i just committed some fixes to master, can you check them out?

thanks

WHAAAT, 2016-05-15:
hello thank you for fast reply.
The master branch fixed the error for qtosgrave plugin.
However fclrave is still the same. I've attached the log for the master branch 
[log.zip](https://github.com/rdiankov/openrave/files/264906/log.log.zip)

rdiankov, 2016-05-16:
it looks like we weren't forcing c++11, can you pull from master and try again?

thanks

WHAAAT, 2016-05-16:
allright! 
There was still a problem with the ros-indigo-fcl package, i had to purge it and install the fcl master branch. Now everything is working fine! Thank you :+1:

hustfzt2, 2016-05-28:
how to solve the problem about " fclrave" in ROS Indigo as:
openrave/plugins/fclrave/fclspace.h:15:9: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
 typedef std::shared_ptr<fcl::CollisionGeometry> CollisionGeometryPtr;

And How to install the fcl master branch ?

rdiankov, 2016-05-29:
i think this compilation problem should be fixed on the latest master

palmieri, 2016-05-31:
Hi all, 

I am having also the same problem with fclrave although I have just git pulled from the master branch. Is the only solution the removal of ros-indigo-fcl package?

Thanls

hustfzt2, 2016-06-01:
hi, all
when ros-indigo-fcl is uninstalled,  the compile for Openrave-latest-stable is successful. However, the simulation GUI for "openrave.py --example hanoi" is not displayed.
How to solve it?

hustfzt2, 2016-06-19:
I am sure that something wrong with ros-indigo-fcl,how to deal with it?

anderwm, 2016-06-29:
I believe ros-indigo-fcl is an older version of fcl that moveit-core depends on.  This is before fcl changed from boost to c++11.  It appears openRave depends on the newer version with c++11 style shared pointers.

robosamir, 2016-07-21:
Workaround: 
Edit your .bashrc to 'unsource' your ros environment and then compile openrave.
This worked successfully for me. removing ros-indigo-fcl was not an option for me since I use moveit.

hustfzt2, 2016-07-23:
however, how to 'unsource' the ros environment?  @robosamir