rdiankov/openrave #402: Inverse dynamics are wrong when actuator attached to last link

stephane-caron, 2016-05-17:
Hi there,

I think there is a bug in the ComputeInverseDynamics() function, checked today on the latest commit on ``master``. The output is wrong when an actuator is attached to the last link (end effector) of a kinematic chain, this way:

    <extra name="LAST_LINK" type="attach_actuator">
        <technique profile="OpenRAVE">
            <instance_actuator url="#actuator18"/>
            <bind_actuator joint="kmodel1/jointsid7"/>
        </technique>
    </extra>

To check the output of ComputeInverseDynamics(), I use the formula

    G = Jcom.T * (robot.mass * gravity)

where Jcom is the COM jacobian and G is the gravity term in the inverse dynamics:

    _, _, G = robot.ComputeInverseDynamics(*, None, returncomponents=True)
    mass = sum([link.GetMass() for link in robot.GetLinks()])
    gravity = env.GetPhysicsEngine().GetGravity()

I checked the formula on various robot models and it works, except on the one with the actuator attached to the last link. When the actuator is present, I get

    norm(G) = 151.437040746
    norm(G - Jcom.T * (robot.mass * gravity)) = 8.20684219593

While if I comment out the actuator the norm goes back to zero as expected:

    norm(G) = 144.560927875  # so, actuators have a dummy mass as well
    norm(G - Jcom.T * (robot.mass * gravity)) = 2.94041472938e-14

No, I don't see the point in having an actuator dangling on the last link, but I didn't make the robot model ;-) However, it's a problem if OpenRAVE returns a wrong result in this case.

rdiankov, 2016-05-20:
stephane,

i see you are starting to play around with more advanced features ;0)

actually all our robots have actuators attached. unfortunately the way openrave prioritizes opening collada files, it puts joints with actuators in front of joints without any actuators (that is how you determine the order of the DOF values).
if you look carefully, you should see that robot.GetJointFromDOFIndex(0) probably returns your last joint.

To fix this: put actuators on all your joints.

good luck!

stephane-caron, 2016-05-20:
Hi Rosen,

Thanks for your answer, that's a precious piece of info to me :+1: 

Unfortunately in this case the problem does not come from the DOF ordering (the relation ``G = J_com^T * mg`` should hold whatever the DOF order is).

Also, I found the guilty actuator in my model does not change DOF ordering when commented/uncommented because the corresponding joint axis is inactive (so there is no DOF corresponding to the joint). From the Collada file:

	<axis_info sid="axis_info_inst6" axis="kmodel1/jointsid7/axis0">
		<active>
			<bool>false</bool>
		</active>
		<locked>
			<bool>false</bool>
		</locked>
		<limits>
			<min>
				<float>0</float>
			</min>
			<max>
				<float>0</float>
			</max>
		</limits>
	</axis_info>

Maybe this case is not taken into account by ``ComputeInverseDynamics``?

rdiankov, 2016-05-23:
we should take into account the passive joints correctly. can you check your link masses? perhaps you have a dummy link somewhere that is contributing to the extra torque?

i'd have to see your model to be of any more help... ;0/

rdiankov, 2016-05-24:
thanks for the test script, i fixed the problem. it was with attaching an actuator to a passive joint... which we didn't have extra checking for, now we do ;0)

closed 2016-05-24 by rdiankov