rdiankov/openrave #405: more recent sympy versions

kb0n, 2016-06-01:
Hi there,

I'm currently in the process of adapting ikfast to more recent sympy versions,  v0.7.6 to be more precise. Not only for using it with Fedora23, but also for the packaging of Openrave for Fedora.

Anyhow, I ended up with a version that works for us. I'm fairly certain though that many scenarios of ikfast were not triggered, so I was wondering if there was any way to thoroughly test ikfast.

I appreciate the help. I'll also create a pull request soonish with my current patches, so you can keep track of the progress, but also for Q&A on aspects of ikfast or other problems I might encounter. If that's ok with you.
I imaging there is a general desire to be compatible with a more recent version rather than being stuck with 0.7.1?

Cheers

rdiankov, 2016-06-03:
Awesome!
By the way, for testing stuff, you can try:

python run_tests.py --test test_ikfast_mandatory

it requires nose testing library.

in sandbox i also have:

python updateikfiles.py --forcegenerate --destdir=../plugins/ikfastsolvers --testnum=100

hope this helps

kb0n, 2016-06-03:
Thanks, I will try those.

Why do you run the second command in a sandbox? Should I be worried if I dont'?

rdiankov, 2016-06-03:
in plugins/ikfastsolvers you'll see there's ikfast generated solutions for some of the samples we have. it's not necessary, but it helps people run openrave without having to wait 5min to genreate the ikfast solution. the updateikfiles.py updates to files.