rdiankov/openrave #410: ikfast: bug in LC-normalization

kb0n, 2016-06-07:
I'm refererring to these lines  of the `checkFinalEquation` method:
```python
[8141]    # since coeffs[0] is normalized with the LC constant, can compare for precision
[8142]    if len(coeffs) == 1 and Abs(coeffs[0]) < 2*(10.0**-self.precision):
```
During some tests with printouts I had
```python
coeffs[0] = 1.0*new_r00*new_r22 - 1.0*new_r11*new_r22**2
```
so it could not be compared, as the expression `"Abs(coeffs[0]) < 2*(10.0**-self.precision)"` was a relation and its boolean value therefore not evaluable. In sympy < 0.7.4 the comparison simply returned `False` and IKFast continued (and finished), but since then you get:
```python
TypeError("symbolic boolean expression has no truth value.")
```

You can easily reproduce this with:
```python
from sympy import *
x = symbols('x')
a = x < 1
bool(a)
```

It would be easy to catch the error and continue with a `False` value (IKFast finishes then), but from what I understand the comparison _should_ be valid (see line 8141). So I assume there is a bug somewhere prior to that when filtering the coeffs, maybe in `removecommonexprs` method, maybe somewhere else. I dont' really know as it is pretty difficult for me to understand all that, but I guess a proper fix would be preferable rather than error-cathing?

rdiankov, 2016-06-07:
yup, the bug is definitely somewhere else! ;0)

the fact that sympy was masking that error is really bad and probably that's why i haven't caught it yet (and it's behavior like that that make using new sympy versions scary, even though they can be more correct ;0/ )

rdiankov, 2016-06-07:
by the way, what robot is this on?

kb0n, 2016-06-07:
haha, I totally get that :D

This error occured for a model of the Neuronics Katana arm with a 6th dummy joint: [link](https://git.fawkesrobotics.org/fawkes.git/blob_plain/HEAD:/src/plugins/openrave/manipulators/katana.kinbody.xml)