rdiankov/openrave #417: problem related to libqtosgrave

palmieri, 2016-07-01:
Hi all,

I have installed openrave from the master branch. The compilation goes smooth but when I try to launch the example "hanoi", I receive the following error.

> openrave.py --example hanoi
2016-07-01 22:23:27,539 openrave [WARN] [plugindatabase.h:929 RaveDatabase::_SysLoadLibrary] /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: /usr/local/lib/openrave0.9-plugins/libqtosgrave.so: undefined symbol: _ZN3osg5Group11removeChildEPNS_4NodeE
[1]    23219 segmentation fault (core dumped)  openrave.py --example hanoi

The version of OpenSceneGraph that I have is the 3.4 (obtained by using git clone --branch OpenSceneGraph-3.4.0 https://github.com/openscenegraph/OpenSceneGraph.git).

Anyway in the CMakeLists.txt file at line 46 I putting to OFF the option OPT_QTOSG_VIEWER  too.

The environment variable LD_LIBRARY_PATH is set so to include the path  where the openscenegraph lib is. 

Any idea of what is happening? Thanks for your help. 

Best,

Luigi

RunOrVeith, 2016-07-12:
I can confirm, it happens also when just starting openrave without an example.
The OPT_QTOSG_VIEWER variable is set to true for me though, all entries are defined in the CMake for OSG.

RunOrVeith, 2016-07-13:
I found a fix if anyone else has this issue:
This seems to happen when more than one version of OpenSceneGraph is installed.
On my system (Ubuntu 14.04), version 3.2. was installed already, but openRAVE needs version 3.4 or newer. It is not enough to simply build the newer version from source, you also need to remove the old version. The names of the installed packages that I had to remove are:
openscenegraph libopenscenegraph-dev libopenscenegraph99

jmkinzer, 2016-08-02:
Thanks for identifying this fix! I just wanted to add  that the libopenscenegraph may vary. In my case (16.4) for example I had to remove libopenscenegraph100v5. Best just to do something like "apt list --installed | grep openscene" and remove everything one finds...

rdiankov, 2016-08-21:
glad you guys resolved this!

closed 2016-08-21 by rdiankov

lakshmip001, 2019-01-15:
on Ubuntu 16.04 , when I remove libopenscenegraph100v5 it also removes libopencenegraph-dev. when try installing libopenscenegraph-dev , it also get installed. Kindly please help me how I can slove.