rdiankov/openrave #419: replaceObject not available in fcl

yn, 2016-07-16:
When compiling current master, I get this:

/openrave/plugins/fclrave/fclmanagercache.h:294:43: error: no member named 'replaceObject' in
      'fcl::BroadPhaseCollisionManager'
                                pmanager->replaceObject(itcache->second.vcolobjs.at(ilink).get(), pcol.get(), false);
                                ~~~~~~~~  ^

I am not sure where replaceObject is supposed to be coming from; it is not present in the current sources at https://github.com/flexible-collision-library/fcl

kyoDralliam, 2016-07-16:
Openrave is currently using a modified version of the fcl library that you can find at https://github.com/rdiankov/fcl/tree/kenjiSpeedUpAdditions

rdiankov, 2016-07-17:
oops, sorry about this!
we'll put in some compiler flags to detect this automatically

until then, please turn off compiling FCL with: -DOPT_FCL_COLLISION=OFF

rdiankov, 2016-07-18:
we just committed CMake fixes for fcl, can you check again?

thanks

yn, 2016-07-18:
The previous error related to replaceObject went away. I have now the following errors (actually, I had them previously, but didn't mention them because I was trying to make the report isolated:

```
[ 99%] Building CXX object plugins/fclrave/CMakeFiles/fclrave.dir/fclrave.cpp.o
In file included from /Users/yn/code/others/openrave/plugins/fclrave/fclrave.cpp:16:
In file included from /Users/yn/code/others/openrave/plugins/fclrave/plugindefs.h:102:
In file included from /usr/local/Cellar/fcl/0.4.0/include/fcl/collision.h:42:
In file included from /usr/local/Cellar/fcl/0.4.0/include/fcl/math/vec_3f.h:43:
/usr/local/Cellar/fcl/0.4.0/include/fcl/math/math_details.h:350:5: warning: 
      'register' storage class specifier is deprecated [-Wdeprecated-register]
    register T tmp = rs[0][1];
    ^~~~~~~~~
/usr/local/Cellar/fcl/0.4.0/include/fcl/math/math_details.h:367:5: warning: 
      'register' storage class specifier is deprecated [-Wdeprecated-register]
    register T inrsdet = 1 / det;
    ^~~~~~~~~
In file included from /Users/yn/code/others/openrave/plugins/fclrave/fclrave.cpp:20:
/Users/yn/code/others/openrave/include/openrave/plugin.h:59:48: warning: 
      'OpenRAVECreateInterface' has C-linkage specified, but returns
      user-defined type 'OpenRAVE::InterfaceBasePtr' (aka
      'shared_ptr<OpenRAVE::InterfaceBase>') which is incompatible with C
      [-Wreturn-type-c-linkage]
OPENRAVE_PLUGIN_API OpenRAVE::InterfaceBasePtr OpenRAVECreateInterface(O...
                                               ^
In file included from /Users/yn/code/others/openrave/plugins/fclrave/fclrave.cpp:16:
In file included from /Users/yn/code/others/openrave/plugins/fclrave/plugindefs.h:56:
In file included from /usr/local/include/boost/make_shared.hpp:15:
In file included from /usr/local/include/boost/smart_ptr/make_shared.hpp:15:
/usr/local/include/boost/smart_ptr/make_shared_object.hpp:254:17: error: no
      matching constructor for initialization of 'fcl::CollisionObject'
    ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );
                ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/yn/code/others/openrave/plugins/fclrave/fclspace.h:242:54: note: in
      instantiation of function template specialization
      'boost::make_shared<fcl::CollisionObject, const
      std::__1::shared_ptr<fcl::CollisionGeometry> &>' requested here
                CollisionObjectPtr pfclcoll = boost::make_shared<fcl::Co...
                                                     ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:151:7: note: 
      candidate constructor (the implicit copy constructor) not viable: no known
      conversion from 'const std::__1::shared_ptr<fcl::CollisionGeometry>' to
      'const fcl::CollisionObject' for 1st argument
class CollisionObject
      ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:154:2: note: 
      candidate constructor not viable: no known conversion from 'const
      std::__1::shared_ptr<fcl::CollisionGeometry>' to 'const
      boost::shared_ptr<CollisionGeometry>' for 1st argument
 CollisionObject(const boost::shared_ptr<CollisionGeometry> &cgeom_) :
 ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:164:3: note: 
      candidate constructor not viable: requires 2 arguments, but 1 was provided
  CollisionObject(const boost::shared_ptr<CollisionGeometry> &cgeom_, co...
  ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:171:3: note: 
      candidate constructor not viable: requires 3 arguments, but 1 was provided
  CollisionObject(const boost::shared_ptr<CollisionGeometry> &cgeom_, co...
  ^
In file included from /Users/yn/code/others/openrave/plugins/fclrave/fclrave.cpp:16:
In file included from /Users/yn/code/others/openrave/plugins/fclrave/plugindefs.h:56:
In file included from /usr/local/include/boost/make_shared.hpp:15:
In file included from /usr/local/include/boost/smart_ptr/make_shared.hpp:15:
/usr/local/include/boost/smart_ptr/make_shared_object.hpp:254:17: error: no
      matching constructor for initialization of 'fcl::CollisionObject'
    ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );
                ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/yn/code/others/openrave/plugins/fclrave/fclspace.h:259:56: note: in
      instantiation of function template specialization
      'boost::make_shared<fcl::CollisionObject,
      std::__1::shared_ptr<fcl::CollisionGeometry> &>' requested here
                CollisionObjectPtr pfclcollBV = boost::make_shared<fcl::...
                                                       ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:151:7: note: 
      candidate constructor (the implicit copy constructor) not viable: no known
      conversion from 'std::__1::shared_ptr<fcl::CollisionGeometry>' to 'const
      fcl::CollisionObject' for 1st argument
class CollisionObject
      ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:154:2: note: 
      candidate constructor not viable: no known conversion from
      'std::__1::shared_ptr<fcl::CollisionGeometry>' to 'const
      boost::shared_ptr<CollisionGeometry>' for 1st argument
 CollisionObject(const boost::shared_ptr<CollisionGeometry> &cgeom_) :
 ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:164:3: note: 
      candidate constructor not viable: requires 2 arguments, but 1 was provided
  CollisionObject(const boost::shared_ptr<CollisionGeometry> &cgeom_, co...
  ^
/usr/local/Cellar/fcl/0.4.0/include/fcl/collision_object.h:171:3: note: 
      candidate constructor not viable: requires 3 arguments, but 1 was provided
  CollisionObject(const boost::shared_ptr<CollisionGeometry> &cgeom_, co...
  ^
3 warnings and 2 errors generated.
make[3]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/fclrave.cpp.o] Error 1
make[2]: *** [plugins/fclrave/CMakeFiles/fclrave.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

yn, 2016-07-18:
FWIW, here are the versions of FCL and BOOST I have installed:

```
∴ brew info fcl boost
dartsim/dart/fcl: stable 0.4.0, HEAD
http://gamma.cs.unc.edu/FCL
/usr/local/Cellar/fcl/0.4.0 (97 files, 8.6M) *
  Built from source on 2016-07-13 at 13:51:48
From: https://github.com/dartsim/homebrew-dart/blob/master/fcl.rb
==> Dependencies
Build: cmake ✘, pkg-config ✘
Required: boost ✔, homebrew/science/libccd ✔
Optional: homebrew/science/flann ✘
==> Options
--with-flann
	Build with flann support
--HEAD
	Install HEAD version

boost: stable 1.60.0 (bottled), HEAD
Collection of portable C++ source libraries
https://www.boost.org/
/usr/local/Cellar/boost/1.55.0_1 (10,036 files, 429.5M)
  Poured from bottle on 2014-05-08 at 06:56:02
/usr/local/Cellar/boost/1.60.0_2 (11,139 files, 436.8M) *
  Poured from bottle on 2016-07-13 at 13:49:10
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost.rb
==> Dependencies
Optional: icu4c ✔
==> Options
--c++11
	Build using C++11 mode
--universal
	Build a universal binary
--with-icu4c
	Build regexp engine with icu support
--with-mpi
	Build with MPI support
--without-single
	Disable building single-threading variant
--without-static
	Disable building static library variant
--HEAD
	Install HEAD version
```

Ericcherish, 2017-03-20:
Is there any installation tutorial for ubuntu16.04 @rdiankov 
There are so much problem for fresh me!

Puttichai, 2017-03-20:
@Ericcherish Try [this](https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html).

Ericcherish, 2017-03-20:
very good, Thank you @Puttichai