rdiankov/openrave #422: Error while compile - ikfastsolver.cpp.o

Softwaretom007, 2016-07-26:
Hi,
I can't compile latest_stable or master branch. Here is error log:
`Building CXX object plugins/ikfastsolvers/CMakeFiles/ikfastsolvers.dir/ikfastsolver.cpp.o
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1991:5: error: ‘ikfastsolvers’ does not name a type
     ikfastsolvers::JacobianInverseSolver<double> _jacobinvsolver; ///< jacobian inverse solver if _fRefineWithJacobianInverseAllowedError is > 0
     ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp: In member function ‘bool IkFastSolver<IkReal>::_SetJacobianRefineCommand(std::ostream&, std::istream&)’:
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:101:9: error: ‘_jacobinvsolver’ was not declared in this scope
         _jacobinvsolver.SetErrorThresh(_fRefineWithJacobianInverseAllowedError);
         ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp: In member function ‘void IkFastSolver<IkReal>::_CheckRefineSolution(const OpenRAVE::IkParameterization&, const OpenRAVE::RobotBase::Manipulator&, std::vector<double>&)’:
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1208:21: error: ‘_jacobinvsolver’ was not declared in this scope
                 if( _jacobinvsolver.ComputeSolution(param.GetTransform6D(), manip, vsolution) == 0 ) {
                     ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp: In instantiation of ‘bool IkFastSolver<IkReal>::_CallIk2(const OpenRAVE::IkParameterization&, const std::vector<T>&, const Transform&, ikfast::IkSolutionList<IkReal>&) [with IkReal = float; OpenRAVE::Transform = OpenRAVE::geometry::RaveTransform<double>]’:
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:831:68:   required from ‘bool IkFastSolver<IkReal>::_CallIk(const OpenRAVE::IkParameterization&, const std::vector<T>&, const Transform&, ikfast::IkSolutionList<IkReal>&) [with IkReal = float; OpenRAVE::Transform = OpenRAVE::geometry::RaveTransform<double>]’
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1124:77:   required from ‘OpenRAVE::IkReturnAction IkFastSolver<IkReal>::_SolveSingle(const OpenRAVE::IkParameterization&, const std::vector<T>&, const std::vector<double>&, int, OpenRAVE::IkReturnPtr, IkFastSolver<IkReal>::StateCheckEndEffector&) [with IkReal = float; OpenRAVE::IkReturnPtr = boost::shared_ptr<OpenRAVE::IkReturn>]’
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:600:92:   required from ‘bool IkFastSolver<IkReal>::Solve(const OpenRAVE::IkParameterization&, const std::vector<double>&, int, OpenRAVE::IkReturnPtr) [with IkReal = float; OpenRAVE::IkReturnPtr = boost::shared_ptr<OpenRAVE::IkReturn>]’
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:2014:1:   required from here
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:985:69: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::trans.OpenRAVE::geometry::RaveVector<double>::x’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
                 IkReal eetrans[3] = {t.trans.x, t.trans.y, t.trans.z};
                                                                     ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:985:69: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::trans.OpenRAVE::geometry::RaveVector<double>::y’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:985:69: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::trans.OpenRAVE::geometry::RaveVector<double>::z’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[0]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
                 IkReal eerot[9] = {t.m[0],t.m[1],t.m[2],t.m[4],t.m[5],t.m[6],t.m[8],t.m[9],t.m[10]};
                                                                                                   ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[1]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[2]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[4]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[5]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[6]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[8]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[9]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:986:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[10]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[0]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
                 IkReal eerot[9] = {t.m[0],t.m[1],t.m[2],t.m[4],t.m[5],t.m[6],t.m[8],t.m[9],t.m[10]};
                                                                                                   ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[1]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[2]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[4]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[5]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[6]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[8]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[9]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1001:99: warning: narrowing conversion of ‘t.OpenRAVE::geometry::RaveTransformMatrix<double>::m[10]’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
[... cut ...]
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:959:68: warning: narrowing conversion of ‘p.std::pair<OpenRAVE::geometry::RaveVector<double>, double>::second’ from ‘double’ to ‘float’ inside { } is ill-formed in C++11 [-Wnarrowing]
                 IkReal eerot[9] = {p.second, 0, 0, 0, 0, 0, 0, 0, 0};
                                                                    ^
make[2]: *** [plugins/ikfastsolvers/CMakeFiles/ikfastsolvers.dir/ikfastsolver.cpp.o] Error 1
make[1]: *** [plugins/ikfastsolvers/CMakeFiles/ikfastsolvers.dir/all] Error 2
make: *** [all] Error 2
`
Someone can help? Thanks

jackros1022, 2016-07-29:
Hi,you did it?

rdiankov, 2016-07-29:
you are probably compiling with floating point? can you try double precision -DOPT_DOUBLE_PRECISION=ON

Softwaretom007, 2016-07-29:
Hi, no, still stuck at this.
Yes I was compiling in single. Now I switch to double, but still same error:
`error: ‘ikfastsolvers’ does not name a type
     ikfastsolvers::JacobianInverseSolver<double> _jacobinvsolver; ///< jacobian inverse solver if _fRefineWithJacobianInverseAllowedError is > 0
     ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp: In member function ‘bool IkFastSolver<IkReal>::_SetJacobianRefineCommand(std::ostream&, std::istream&)’:
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:101:9: error: ‘_jacobinvsolver’ was not declared in this scope
         _jacobinvsolver.SetErrorThresh(_fRefineWithJacobianInverseAllowedError);
         ^
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp: In member function ‘void IkFastSolver<IkReal>::_CheckRefineSolution(const OpenRAVE::IkParameterization&, const OpenRAVE::RobotBase::Manipulator&, std::vector<double>&)’:
/home/pc-sedaky/openrave/plugins/ikfastsolvers/ikfastsolver.cpp:1208:21: error: ‘_jacobinvsolver’ was not declared in this scope
`

rdiankov, 2016-07-29:
ah, you are right. there is a problem. you were missing liblapack-dev and there were several places we forgot to double check. i just committed the changes to master branch, but you can also install liblapack-dev.
thanks

closed 2016-07-29 by rdiankov