rdiankov/openrave #425: cmake ..  appears Error.                  Help me,please!

jackros1022, 2016-07-29:
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Compiling OpenRAVE Version 0.9.0, soversion=0.9
-- Using cmake version 2.8.12
-- Using double precision
-- detected system processor: x86_64
**CMake Error at CMakeLists.txt:84 (include):**
  include could not find load file:

    FindIntl

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.6", minimum required is "2") 
-- installing to /usr/local, python exe /usr/bin/python, python subdir is lib/python2.7/dist-packages, program suffix is '0.9'
-- Performing Test LINKER_HAS_RDYNAMIC
-- Performing Test LINKER_HAS_RDYNAMIC - Success
-- Performing Test LINKER_HAS_BSYMBOLIC
-- Performing Test LINKER_HAS_BSYMBOLIC - Success
-- Performing Test LINKER_HAS_BSYMBOLIC_FUNCTIONS
-- Performing Test LINKER_HAS_BSYMBOLIC_FUNCTIONS - Success
-- Performing Test LINKER_HAS_VISIBILITY
-- Performing Test LINKER_HAS_VISIBILITY - Success
-- Performing Test LINKER_HAS_VISIBILITY_INLINES_HIDDEN
-- Performing Test LINKER_HAS_VISIBILITY_INLINES_HIDDEN - Success
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   regex
--   filesystem
--   system
--   python
--   thread
--   iostreams
--   date_time
-- found boost version: 105400
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmp.so
-- Found GMPXX: /usr/lib/x86_64-linux-gnu/libgmpxx.so
-- Found MPFR: /usr/lib/x86_64-linux-gnu/libmpfr.so
-- Looking for mpfr_div_d in /usr/lib/x86_64-linux-gnu/libmpfr.so
-- Looking for mpfr_div_d in /usr/lib/x86_64-linux-gnu/libmpfr.so - found
-- Could NOT find MPFI
-- Looking for crlibm_init in crlibm
-- Looking for crlibm_init in crlibm - not found
-- Looking for crlibm.h
-- Looking for crlibm.h - not found
-- Using local crlibm
-- Looking for fpu_control.h
-- Looking for fpu_control.h - found
-- Looking for fenv.h
-- Looking for fenv.h - found
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Check size of long double
-- Check size of long double - done
-- compiling with double extended (long double) floating point
**CMake Warning at CMakeLists.txt:479 (find_package):**
  By not providing "FindASSIMP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ASSIMP", but
  CMake did not find one.

  Could not find a package configuration file provided by "ASSIMP" with any
  of the following names:

    ASSIMPConfig.cmake
    assimp-config.cmake

  Add the installation prefix of "ASSIMP" to CMAKE_PREFIX_PATH or set
  "ASSIMP_DIR" to a directory containing one of the above files.  If "ASSIMP"
  provides a separate development package or SDK, be sure it has been
  installed.

-- checking for module 'assimp'
--   found assimp, version 3.0.1264
-- assimp version 3.0.1264, , /usr/include/assimp
-- Performing Test IS_ASSIMP_PRE_R896
-- Performing Test IS_ASSIMP_PRE_R896 - Failed
-- Performing Test IS_X86_64
-- Performing Test IS_X86_64 - Success
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- WARNING: you are using the obsolete 'PKGCONFIG' macro, use FindPkgConfig
-- Found LibXml2: /usr/lib/x86_64-linux-gnu/libxml2.so
-- libxml2 found
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake-qt4 (found version "4.8.6") 
-- checking for module 'libpcrecpp'
--   found libpcrecpp, version 8.31
-- Looking for C++ include pcrecpp.h
-- Looking for C++ include pcrecpp.h - found
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- checking for module 'minizip'
--   package 'minizip' not found
-- Looking for qh_new_qhull in qhull
-- Looking for qh_new_qhull in qhull - found
-- Looking for qhull/qhull_a.h
-- Looking for qhull/qhull_a.h - found
-- Looking for C++ include ANN/ANN.h
-- Looking for C++ include ANN/ANN.h - not found
-- compiling local ann library
**CMake Warning at CMakeLists.txt:710 (find_package):**
  By not providing "FindFPARSER.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "FPARSER", but
  CMake did not find one.

  Could not find a package configuration file provided by "FPARSER"
  (requested version 4.4.3) with any of the following names:

    FPARSERConfig.cmake
    fparser-config.cmake

  Add the installation prefix of "FPARSER" to CMAKE_PREFIX_PATH or set
  "FPARSER_DIR" to a directory containing one of the above files.  If
  "FPARSER" provides a separate development package or SDK, be sure it has
  been installed.

-- checking for one of the modules 'fparser'
-- compiling local fparser library
-- Check size of float
-- Check size of float - done
-- Check size of long int
-- Check size of long int - done
-- Looking for alloca.h
-- Looking for alloca.h - found
-- Looking for strtof
-- Looking for strtof - found
-- Looking for strtold
-- Looking for strtold - found
-- Check size of long double
-- Check size of long double - done
-- checking for module 'liblog4cxx'
--   found liblog4cxx, version 0.10.0
-- compiling local convexdecomposition library
-- Looking for C++ include ext/hash_map
-- Looking for C++ include ext/hash_map - found
-- Looking for asinh
-- Looking for asinh - not found
-- Looking for acosh
-- Looking for acosh - not found
-- Looking for atanh
-- Looking for atanh - not found
-- Looking for exp2
-- Looking for exp2 - not found
-- Looking for log2
-- Looking for log2 - not found
-- Looking for mkstemp
-- Looking for mkstemp - found
-- Geometry Parsing: Collada DOM 2.4.4
-- Geometry Parsing: Using Open Asset Import Library 3.0.1264
-- Geometry Parsing: Using ivcon
-- extracting models to /home/jack/openrave/src
-- Looking for _init in lapack
-- Looking for _init in lapack - found
-- MATLAB installation not found, is mex in the system path?
-- Octave installation not found, is 'mkoctfile' in the system path?
-- Found sympy version '0.7.4.1'
-- Assuming sympy version 0.7.x or greater
-- Performing Test HAVE_ALL_PYTHON_HEADERS
-- Performing Test HAVE_ALL_PYTHON_HEADERS - Success
-- python and boost-python found
-- Looking for cgesdd_ in lapack
-- Looking for cgesdd_ in lapack - found
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found osgDB: /usr/lib/libosgDB.so  
-- Found osgQt: /usr/lib/libosgQt.so  
-- Found osgGA: /usr/lib/libosgGA.so  
-- Found osgFX: /usr/lib/libosgFX.so  
-- Found osgText: /usr/lib/libosgText.so  
-- Found osgViewer: /usr/lib/libosgViewer.so  
-- Found osgManipulator: /usr/lib/libosgManipulator.so  
-- Found osg: /usr/lib/libosg.so  
-- Found OpenThreads: /usr/lib/libOpenThreads.so  
-- Could NOT find OpenSceneGraph: Found unsuitable version "3.2.0", but required is at least "3.4" (found /usr/lib/libosgDB.so;/usr/lib/libosgQt.so;/usr/lib/libosgGA.so;/usr/lib/libosgFX.so;/usr/lib/libosgText.so;/usr/lib/libosgViewer.so;/usr/lib/libosgManipulator.so;/usr/lib/libosg.so;/usr/lib/libOpenThreads.so)
-- Could not find OpenSceneGraph v3+. Please install OSG (http://www.openscenegraph.org/projects/osg)
-- Detected SoQt/Coin3D GUI, making plugin
-- Looking for X11/Xlib.h
-- Looking for X11/Xlib.h - found
-- Looking for dAllocateODEDataForThread
-- Looking for dAllocateODEDataForThread - found
-- Detected ODE multi-threading extensions, but not sure if should enable them... define ODE_USE_MULTITHREAD variable to use them
-- Performing Test NEED_TRIINDEX
-- Performing Test NEED_TRIINDEX - Failed
-- ODE found, building oderave plugin, precision=DOUBLE
-- Configuring incomplete, errors occurred!
See also "/home/jack/openrave/build/CMakeFiles/CMakeOutput.log".
See also "/home/jack/openrave/build/CMakeFiles/CMakeError.log".

rdiankov, 2016-07-29:
interesting........ why wouldn't you have FindIntl...?
i just removed it and commited to master, can you double check?

thanks

closed 2016-07-29 by rdiankov

vlthrasher, 2017-06-21:
I am having this same issue- any hint to what ended up working?

kashif1973, 2023-08-26:
is my issue also relates to this problem
D:\planner\omplapp\build>cmake .. -DCMAKE_TOOLCHAIN_FILE=C:/src/vcpkg/scripts/buildsystems/vcpkg.cmake
-- Build type: Release
-- Could NOT find PY_OpenGL (missing: PY_OPENGL)
CMake Warning at CMakeLists.txt:106 (message):
  Both OpenGL and the Python OpenGL module need to be installed to use the
  GUI

-- Checking for module 'flann'
--   Package 'flann', required by 'virtual:world', not found
-- Checking for module 'libspot'
--   Package 'libspot', required by 'virtual:world', not found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
CMake Error at C:/src/vcpkg/scripts/buildsystems/vcpkg.cmake:855 (_find_package):
  By not providing "Findassimp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "assimp", but
  CMake did not find one.

  Could not find a package configuration file provided by "assimp" with any
  of the following names:

    assimpConfig.cmake
    assimp-config.cmake

  Add the installation prefix of "assimp" to CMAKE_PREFIX_PATH or set
  "assimp_DIR" to a directory containing one of the above files.  If "assimp"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:127 (find_package)

-- Configuring incomplete, errors occurred!

D:\planner\omplapp\build>cmake .. -DVCPKG_TARGET_TRIPLET=x64-windows -DCMAKE_TOOLCHAIN_FILE=C:/src/vcpkg/scripts/buildsystems/vcpkg.cmake
-- Build type: Release
-- Could NOT find PY_OpenGL (missing: PY_OPENGL)
CMake Warning at CMakeLists.txt:106 (message):
  Both OpenGL and the Python OpenGL module need to be installed to use the
  GUI

-- Checking for module 'flann'
--   Package 'flann', required by 'virtual:world', not found
-- Checking for module 'libspot'
--   Package 'libspot', required by 'virtual:world', not found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
CMake Error at C:/src/vcpkg/scripts/buildsystems/vcpkg.cmake:855 (_find_package):
  By not providing "Findassimp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "assimp", but
  CMake did not find one.

  Could not find a package configuration file provided by "assimp" with any
  of the following names:

    assimpConfig.cmake
    assimp-config.cmake

  Add the installation prefix of "assimp" to CMAKE_PREFIX_PATH or set
  "assimp_DIR" to a directory containing one of the above files.  If "assimp"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:127 (find_package)

-- Configuring incomplete, errors occurred!

cielavenir, 2023-08-28:
you just need to install https://github.com/assimp/assimp/