rdiankov/openrave #428: test failures and errors

morxa, 2016-08-03:
When I run the test suite (`python test/run_tests.py`), many test fail or produce errors:

    Ran 157 tests in 1401.812s
    
    FAILED (errors=32, failures=19)

See my [results.xml](https://thofmann.fedorapeople.org/openrave/results.xml) for full results. These results are with an older version (8bfb8a6 to be exact), but it looks similar for the current master.

Is this expected, or should all tests succeed in the current master branch?

Also, I noticed that there are differences depending on whether I installed openrave or not, even if I set PYTHONPATH, LD_LIBRARY_PATH, OPENRAVE_MODELS, OPENRAVE_DATA and PATH to the correct paths. It looks like at least once the test suite calls `openrave-config --shared`, which of course produces the wrong path if openrave isn't installed. Is running the test suite supported without installing openrave first?

We're currently debating if it makes sense to run the test suite during the build process for Fedora, see [here](https://bugzilla.redhat.com/show_bug.cgi?id=1338339) for more information.

morxa, 2016-09-23:
[full build log](https://thofmann.fedorapeople.org/openrave/build.log)

morxa, 2016-11-04:
Any update on this? It would be nice to finalize the package.

morxa, 2017-03-17:
Same issue with latest commit  6607fef:

    Ran 162 tests in 431.342s
    FAILED (errors=34, failures=30)

See [this build log](https://thofmann.fedorapeople.org/openrave/build-18440625.log).

morxa, 2017-10-11:
ping @rdiankov - not sure if you get notifications for this otherwise.

If you're not interested in having OpenRAVE in Fedora (which is my impression by now), please let me know and I'll stop bugging you.

rdiankov, 2017-10-11:
Hey,
Sorry for the late response.

You are getting failures, that shouldn't be there. When running the tests, are you calling "source openrave.bash" after installing openrave? For now, I think this is the only way to make sure to get the right paths setup.

```
ImportError (libconfigurationcache.so: cannot open shared object file: No such file or directory) (<class 'nose.case.Test'>)
```

libconfigurationcache.so is a plugin that provides a class that is then wrapped in a python module. i'm guessing when the python module is imported, it is not finding this (not in path?)

```
noseplugins.multiprocess 2016-09-23 08:50:03,621: DEBUG batch result is ('test_collada.TestCOLLADA.test_external_extrainfo ... ok\n', 1, [], [], {})
test_collada.TestCOLLADA.test_external_extrainfo ... ok
noseplugins.multiprocess 2016-09-23 08:50:03,622: DEBUG Ran 1 tests (total: 9)
noseplugins.multiprocess 2016-09-23 08:50:03,622: DEBUG joining worker 0
[0;33;48m[plugindatabase.h:395 Init] [0;38;48m[0;33;48m/usr/lib64/openrave/plugins doesn't exist
[0;33;48m[libopenrave.cpp:909 _UpdateDataDirs] [0;38;48m[0;33;48m/usr/share/openrave doesn't exist
[0;33;48m[plugindatabase.h:577 Create] [0;38;48m[0;33;48mFailed to create name fcl_, interface collisionchecker
[0;38;48m[0;33;48m[plugindatabase.h:577 Create] [0;38;48m[0;33;48mFailed to create name fcl_, interface collisionchecker
[0;38;48m[0;31;48m[colladareader.cpp:3719 handleError] [0;38;48m[0;31;48mCOLLADA error: Document is empty
[0;31;48m[colladareader.cpp:3719 handleError] [0;38;48m[0;31;48mCOLLADA error: Error parsing XML in daeLIBXMLPlugin::read
[0;33;48m[colladawriter.cpp:707 _WriteKinBodyExternal] [0;38;48m[0;33;48mdo not support jointlimit writing
[0;38;48m[0;33;48m[colladawriter.cpp:710 _WriteKinBodyExternal] [0;38;48m[0;33;48mdo not support jointweight writing
[0;38;48m[0;33;48m[plugindatabase.h:577 Create] [0;38;48m[0;33;48mFailed to create name fcl_, interface collisionchecker
[0;38;48m[0;31;48m[colladareader.cpp:3719 handleError] [0;38;48m[0;31;48mCOLLADA error: Document is empty
[0;31;48m[colladareader.cpp:3719 handleError] [0;38;48m[0;31;48mCOLLADA error: Error parsing XML in daeLIBXMLPlugin::read
[0;33;48m[colladareader.cpp:4214 _getJointFromRef] [0;38;48m[0;33;48mcould not find collada joint 'kmodel1/joint6'!
[0;33;48m[colladareader.cpp:2908 ExtractRobotManipulators] [0;38;48m[0;33;48mcould not find manipulator 'mynewmanip' gripper joint 'kmodel1/joint6'
noseplugins.multiprocess 2016-09-23 08:50:03,673: DEBUG starting new process on worker 0
```

```
[0;38;48mnoseplugins.multiprocess 2016-09-23 08:50:10,661: DEBUG starting new process on worker 2
openravepytest.TestCOLLADA.test_collada_loading (pid=3064): checking collada robots/barrettsegway.robot.xml
openravepytest.TestCOLLADA.LoadRobot (pid=3064): ('robots/barrettsegway.robot.xml',), {}
noseplugins.multiprocess 2016-09-23 08:50:10,672: DEBUG Waiting for results (27/158 tasks), next timeout=986.098s
noseplugins.multiprocess 2016-09-23 08:50:10,673: DEBUG Results received for worker 0 , /builddir/build/BUILD/openrave-8bfb8a6a96d5606b663436495983b0605f251f60/test/test_collada.py:TestCOLLADA.test_externalref_joints(), new tasks: 0
noseplugins.multiprocess 2016-09-23 08:50:10,674: DEBUG batch result is ('test_collada.TestCOLLADA.test_externalref_joints ... ERROR\n', 1, [], [(<noseplugins.multiprocess.TestLet instance at 0x7fac525cf440>, 'Traceback (most recent call last):\n  File "/usr/lib/python2.7/site-packages/nose/case.py", line 197, in runTest\n    self.test(*self.arg)\n  File "/builddir/build/BUILD/openrave-8bfb8a6a96d5606b663436495983b0605f251f60/test/test_collada.py", line 265, in test_externalref_joints\n    robot=env.GetRobots()[0]\nIndexError: list index out of range\n')], {})
test_collada.TestCOLLADA.test_externalref_joints ... ERROR
```

Perhaps the collada-dom library you are using is bad? Are you using latest master from https://github.com/rdiankov/collada-dom ?

```
noseplugins.multiprocess 2016-09-23 08:50:26,346: DEBUG batch result is ('test_environment.TestEnvironment.test_load_cwd ... FAIL\n', 1, [(<noseplugins.multiprocess.TestLet instance at 0x7fac525cfbd8>, 'Traceback (most recent call last):\n  File "/usr/lib/python2.7/site-packages/nose/case.py", line 197, in runTest\n    self.test(*self.arg)\n  File "/builddir/build/BUILD/openrave-8bfb8a6a96d5606b663436495983b0605f251f60/test/test_environment.py", line 240, in test_load_cwd\n    assert(env.Load(\'testdata/box0.dae\'))\nAssertionError: \n>>  assert(RaveGetEnvironment(1).Load(\'testdata/box0.dae\'))\n    assert(not RaveGetEnvironment(1).Load(\'box0.dae\'))\n    <module \'os\' from \'/usr/lib64/python2.7/os.pyc\'>.chdir(\'testdata\')\n    \n')], [], {})
test_environment.TestEnvironment.test_load_cwd ... FAIL
```
```
noseplugins.multiprocess 2016-09-23 08:50:33,374: DEBUG Results received for worker 1 , /builddir/build/BUILD/openrave-8bfb8a6a96d5606b663436495983b0605f251f60/test/test_environment.py:TestEnvironment.test_unicode(), new tasks: 0
noseplugins.multiprocess 2016-09-23 08:50:33,375: DEBUG batch result is ('test_environment.TestEnvironment.test_unicode ... ERROR\n', 1, [], [(<noseplugins.multiprocess.TestLet instance at 0x7fac525d4200>, 'Traceback (most recent call last):\n  File "/usr/lib/python2.7/site-packages/nose/case.py", line 197, in runTest\n    self.test(*self.arg)\n  File "/builddir/build/BUILD/openrave-8bfb8a6a96d5606b663436495983b0605f251f60/test/test_environment.py", line 98, [... cut ...] ::type boost::shared_ptr<T>::operator->() const [with T = openravepy::PyInterfaceBase; typename boost::detail::sp_member_access<T>::type = openravepy::PyInterfaceBase*], expr: px != 0\n')], {})
```

not loading the robot, which shouldn't happen.

That being said, right now even on my PC not all the tests pass for legit reasons: the tests are too old and the API changed.
It's up to you whether you want to add openrave in Fedora like this, as long as it useful in this state, then go for it.

Best,

morxa, 2017-10-13:
Thank you for your reply!

> You are getting failures, that shouldn't be there. When running the tests, are you calling "source openrave.bash" after installing openrave? For now, I think this is the only way to make sure to get the right paths setup.

Yes and no. These tests are run while building the package, at which stage the files are not in their final locations. So for example the binaries are in `$BUILDROOT/usr/bin` and not in `/usr/bin`. I set up the environment following `openrave.bash`:

```
export PYTHONPATH=%{buildroot}/%{python2_sitearch}                              
export LD_LIBRARY_PATH=%{buildroot}/%{_libdir}                                  
export OPENRAVE_DATA=%{buildroot}/usr/share/openrave                            
export OPENRAVE_PLUGINS=%{buildroot}/%{_libdir}/openrave/plugins                
export OCTAVE_PATH=%{buildroot}/%{_libdir}/octave/packages/openrave-%{version}
```

This should set up all paths correctly, I guess? I forgot the `OCTAVE_PATH` in the previous test run, [here](https://thofmann.fedorapeople.org/openrave/test.log) is an updated log file of just running the tests manually.

> ImportError (libconfigurationcache.so: cannot open shared object file: No such file or directory) (<class 'nose.case.Test'>)
>
> libconfigurationcache.so is a plugin that provides a class that is then [...]

Hm, so `libconfigurationcache.so` is in the path set by `$OPENRAVE_PLUGINS`. Maybe the environment variable is not evaluated correctly somewhere? Later in the log, it says:
```
2017-10-13 11:30:25,627 openrave [DEBUG] [plugindatabase.h:116 RaveDatabase::Plugin::Destroy] RaveDatabase: closing plugin /home/thofmann/rpmbuild/BUILDROOT/openrave-0.9.0-1.fc26.x86_64/usr/lib64/openrave/plugins/libconfigurationcache.so
```
So it seems like it's not a PATH issue.

> Perhaps the collada-dom library you are using is bad? Are you using latest master from https://github.com/rdiankov/collada-dom ?

Um no, I'm using the latest release 2.4.4. Is 2.5 final already? I can update collada-dom in Fedora if it is - it would be awesome if you could tag it as release.
> ```
> noseplugins.multiprocess 2016-09-23 08:50:26,346: DEBUG batch result is ('test_environment.TestEnvironment.test_load_cwd ... FAIL\n', 1, [(<noseplugins.multiprocess.TestLet instance at [...]

Missing $PATH in the environment setup (as above, `openrave-config` is in `$BUILDROOT/usr/bin`), fixed.

> ```
> noseplugins.multiprocess 2016-09-23 08:50:34,267: DEBUG Worker 1 ending
> noseplugins.multiprocess 2016-09-23 08:50:34,268: DEBUG Results received for worker 1 , [...]

Maybe that's because I didn't rebase the patch from #407 ? I'm not sure if that patch is still required.

> 
> That being said, right now even on my PC not all the tests pass for legit reasons: the tests are too old and the API changed.
> It's up to you whether you want to add openrave in Fedora like this, as long as it useful in this state, then go for it.

OK, thanks, I can work with that. I still want to add openrave as it is, I just need to know why the tests fail. If some of them are expected to fail, that's okay.