rdiankov/openrave #429: RuntimeError: tr1::bad_weak_ptr

atenpas, 2016-08-05:
I've created an ikfast solver for a .dae file as a shared library with both the newest _master_ and _latest_stable_ branches of _openrave_. However, when trying to use the ik-solver in Python, I'm getting the following error:

[[-0.78129155 -0.62404095  0.01250626  0.5361106 ]
 [-0.62281184  0.77812039 -0.0814498   0.13360108]
 [ 0.04109663 -0.07142509 -0.99659898 -0.01161083]
 [ 0.          0.          0.          1.        ]]
2016-08-05 13:18:51,089 openrave [ERROR] [ikfastsolver.cpp:239 Init] inverse kinematics hashes do not match for manip baxter:left_arm. IK will not work! 066a871837ad268b50406eb32c2e6dba!=<robot:GenericRobot - baxter (10dd3e922d483bc7a3d64730a1d8add3)>
Traceback (most recent call last):
  File "ikfast_test.py", line 27, in <module>
    sol = manip.FindIKSolution(goal_pose, 18) # get collision-free solution
RuntimeError: tr1::bad_weak_ptr

This did not happen with previous versions of _latest_stable_. When compiling the ikfast cpp-file into an executable, I can use it without problems from the terminal. It then prints multiple solutions for a Cartesian pose input as command line arguments.

I've attached a minimal example to this post.

[baxter_left.tar.gz](https://github.com/rdiankov/openrave/files/404281/baxter_left.tar.gz)

atenpas, 2016-08-05:
As a follow-up. I have managed to access the _ikfast_ shared library through Python's _ctypes_ package. This way, I could solve inverse kinematics without problems. This seems to suggest that the problem lies within the **openrave** source code.

rdiankov, 2016-08-06:
this is not a problem with openrave source code, the new openrave has a different way of computing hashes, which means you have to regenerate the IK, or in your IK file you have to change the returned string of GetKinematicsHash() or you just need to regenerete the IK.

closed 2016-08-06 by rdiankov

atenpas, 2016-08-06:
I fixed it manually in my IK file and it works. 

However, I don't understand why it has the wrong hashes as I did regenerate the ikfast solver on each computer on which I tried this.