rdiankov/openrave #435: Crazy models when env.Load() is called after env.SetViewer()

stephane-caron, 2016-08-18:
Hi guys,

I'm getting a weird behavior when ``env.Load('my_model.dae')`` is called after ``env.SetViewer()``.

If I call it **before**, OpenRAVE loads my model ([JVRC-1.dae](https://github.com/stephane-caron/jvrc_models/blob/master/JVRC-1/JVRC-1.dae)) fine:

[image]

But if I load it **after** doing ``env.SetViewer('qtcoin')`` or ``env.SetViewer('qtosg')``, I get a crazy geometry:

[image]

Any idea what could be causing this?

jgvictores, 2020-01-25:
Just to centralize on this issue that appears a lot:
```bash
openrave robots/mitsubishi-pa10.zae # fails
openrave.py robots/mitsubishi-pa10.zae # works
```

jgvictores, 2020-01-25:
> openrave robots/mitsubishi-pa10.zae # fails
> openrave.py robots/mitsubishi-pa10.zae # works

PS: Just confirmed that the same effect (race-condition or whatever) on Ubuntu 16.04 happens with `qtosg` (the openscenegraph viewer plugin) keeps on happening.

inaciose, 2020-06-21:
2020 tentatives with ubuntu, 14.04, 16.04, and 18.04, the strange geometry still present in all. The hanoi example always crash, after present some errors.

openrave.py --example hanoi
2020-06-21 13:15:10,621 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:10,622 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:10,622 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:10,622 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:10,622 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:10,623 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:10,693 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:10,694 openrave [WARN] [colladareader.cpp:2022 KinBody::LinkPtr OpenRAVE::ColladaReader::ExtractLink] bad joint resolution 0.000000
2020-06-21 13:15:11,215 openrave [ERROR] [ikfastsolver.cpp:256 Init] inverse kinematics hashes do not match for manip Puma:arm. IK will not work! 74458059fde60df7401cc1dfcfa06d15!=a6c70e6dd694838553470dde754d5825
robot resolutions:  [  7.44891743e-05   8.76348043e-05   1.03408517e-04   1.22876902e-04
   1.38651137e-04   1.65434356e-04   2.20197324e-04]
robot weights:  [ 1.55474929  1.32152963  1.11994634  0.94250415  0.83527617  0.700048
  0.52594641]
hanoimove disk2 from srcpeg to destpeg
2020-06-21 13:15:11,261 openrave [INFO] [?:-1 ?] opening hand
2020-06-21 13:15:11,528 openrave [WARN] [basemanipulation.cpp:955 MoveUnsyncJoints] failed to jitter robot out of collision
2020-06-21 13:15:11,619 openrave [WARN] [?:-1 ?] Planning Error
MoveUnsyncJoints
2020-06-21 13:15:11,849 openrave [WARN] [basemanipulation.cpp:955 MoveUnsyncJoints] failed to jitter robot out of collision
Traceback (most recent call last):
  File "/usr/local/bin/openrave.py", line 149, in <module>
    example.run(args=args)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/openravepy_ext.py", line 64, in newfn
    return fn(*args,**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 235, in run
    OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 242, in parseAndCreateThreadedUser
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 102, in SetViewerUserThread
    userfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/misc.py", line 242, in <lambda>
    SetViewerUserThread(env,viewername,lambda: userfn(env,options))
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 215, in main
    hanoi.hanoisolve(3,hanoi.srcpeg,hanoi.destpeg,hanoi.peg)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 205, in hanoisolve
    self.hanoisolve(n-1, pegfrom, pegby, pegto)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 205, in hanoisolve
    self.hanoisolve(n-1, pegfrom, pegby, pegto)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 199, in hanoisolve
    if not self.hanoimove(disk, pegfrom, pegto, self.heights[len(pegto.disks)]):
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 189, in hanoimove
    openhandfn()
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 158, in <lambda>
    openhandfn = lambda: self.MoveToPosition([-0.7],self.robot.GetActiveManipulator().GetGripperIndices())
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/examples/hanoi.py", line 84, in MoveToPosition
    self.basemanip.MoveUnsyncJoints(jointvalues=values,jointinds=indices)
  File "/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/interfaces/BaseManipulation.py", line 253, in MoveUnsyncJoints
    raise PlanningError('MoveUnsyncJoints')
openravepy._openravepy_0_9.openravepy_ext.PlanningError: Planning Error
MoveUnsyncJoints

jgvictores, 2020-06-22:
Hi @inaciose,

From what I recall from what we've been seeing, in the case of Collada (`.dae` and `.zae` models), it all boils down to:
- Works: first load the model (`env.Load('my_model.dae')`), and then set the viewer (`env.SetViewer()`).
- Fails: first set the viewer, then load the model.

To find a working version under this assumption, the process would be to review the code internals to see in which order these functions are called.

inaciose, 2020-06-22:
@jgvictores 

Thankyou for your help. Some info on how I manage to run the hanoi example.

To run the example i use the following line: openrave.py --example hanoi

So I check the loader openrave.py and the hanoi example python code.
The openrave.py have the env.SetViewer() in the code, but I belive that when we run the example it is not executed (I used some caveman debug prints to do some checks)

So I take some time to do a deeper check on the hanoi.py
https://github.com/rdiankov/openrave/blob/master/python/examples/hanoi.py

As i did not see any call env.SetViewer() on hanoi.py the promissing statement was:

OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=True)

Then after following the statement executation I found that we need to change the statement to

OpenRAVEGlobalArguments.parseAndCreateThreadedUser(options,main,defaultviewer=False)

And just after;
env.Load(options.scene) 
Add:
env.SetViewer("qtcoin")
 
Thank you again for your help!