rdiankov/openrave #443: qtosg viewer bug: resizing window

Puttichai, 2016-09-08:
Resizing the viewer's window (vertically) to zero height and resizing the window back causes the program to crash with warnings like:

```
Warning:: Picked up error in TriangleIntersect
   (0 0 -0.00397,	0.00198168 0.00388924 -0.00397,	0.00256569 0.00353135 -0.00397)
   (-nan,	-nan,	-nan)
```

Here is a minimal working example in python:

```python
import openravepy as orpy
env = orpy.Environment()
env.Load("data/lego2.kinbody.xml")
env.SetViewer("qtosg")
```
Then resize the window vertically until it has zero height and resize it back. Warnings will be flushing on python terminal.

Note1: the same behavior also occurs when we do resizing before loading a model
```python
import openravepy as orpy
env = orpy.Environment()
env.SetViewer("qtosg")
# Do resizing here then load a model
env.Load("data/lego2.kinbody.xml")
```

Note2: Warnings stop flushing when pressing `View Model` button in the viewer. Clicking other buttons do not stop the warnings, though.

ziyan, 2016-09-08:
Could be somewhere there is a divide by zero that cause the camera parameters to have NaN in it.

Puttichai, 2016-09-08:
I think so. May be [this line](https://github.com/rdiankov/openrave/blob/master/plugins/qtosgrave/qtosgviewer.cpp#L1065) in `qtosgviewer.cpp`.