rdiankov/openrave #445: fcl_ collision detector ignore primitives?

Softwaretom007, 2016-09-13:
Hello, i have an issue with fcl collision detector. Everything is fine when i use "pqp". But its slow, so i want to use "fcl_". Here is .mp4 video (please, change txt to mp4).
[openrave_vid.txt](https://github.com/rdiankov/openrave/files/469795/openrave_vid.txt)
But robot themself is not ignoring collision with primitives (eg. walls or obstacles). I think only attached (grasped) object is problem. But I am not sure about this.
When i switch to ode, its crashed with error: 

> python: symbol lookup error: /usr/local/lib/libode.so.6: undefined symbol: _ZN5odeou18CTLSInitialization16InitializeTLSAPIERNS_18CTypeSimpleWrapperIPNS1_IPvLi1EEELi0EEEjj

Thanks for response

rdiankov, 2016-09-14:
wow, that's bad....

for fcl, what  is the exact version you are compiling on? we use a local modified version, so that mimght be the problem.

for ODE, it seems you have multiple libraries installed?

Softwaretom007, 2016-09-14:
Hi, rdiankov.
Yes, in this case is this collision checker absolutely useless.
I have this fcl_ : https://github.com/rdiankov/fcl  ,branch master. FCL implemented in openrave package I cannot compile. 
Openrave is from git too, branch master, approx 2 months old.
For ODE, I use https://sourceforge.net/projects/opende/ newest version i think. I will try uninstall it and recompile.

cielavenir, 2018-01-18:
I think fcl grabbed body issue was fixed in https://github.com/rdiankov/openrave/commit/81079b82442ac83fe212dc0453ed849d6c8aea2c

N3k4rk, 2018-08-10:
I had the same problem,  but as of august 2018, the fix mentioned by cielavenir doesnt seem to be in either master (Latest commit 7c5f5e2 on Feb 19) or latest_stable (Latest commit 9c79ea2 on Aug 13, 2016) branch. I have manually edited the file mentioned and after rebuilding openrave it seems to have fixed the problem. Im using fcl_  0.5.0 from  https://github.com/flexible-collision-library/fcl