rdiankov/openrave #458: Unable to install from Source on Ubuntu 16.04.1

ritwik1993, 2016-11-08:
I am currently unable to build openrave(master branch) on my Ubuntu system.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
Kernel release- 4.4.0-36-generic #55-Ubuntu SMP

It fails at the following step - 
```
CMakeFiles/openrave.dir/openrave.cpp.o: In function `OpenRAVE::RavePrintfA_WARNLEVEL(log4cxx::helpers::ObjectPtrT<log4cxx::Logger> const&, log4cxx::spi::LocationInfo const&, char const*, ...)':
openrave.cpp:(.text._ZN8OpenRAVE21RavePrintfA_WARNLEVELERKN7log4cxx7helpers10ObjectPtrTINS0_6LoggerEEERKNS0_3spi12LocationInfoEPKcz[_ZN8OpenRAVE21RavePrintfA_WARNLEVELERKN7log4cxx7helpers10ObjectPtrTINS0_6LoggerEEERKNS0_3spi12LocationInfoEPKcz]+0x28e): undefined reference to `log4cxx::Logger::forcedLog(log4cxx::helpers::ObjectPtrT<log4cxx::Level> const&, std::string const&, log4cxx::spi::LocationInfo const&) const'
CMakeFiles/openrave.dir/openrave.cpp.o: In function `OpenRAVE::RavePrintfA_DEBUGLEVEL(log4cxx::helpers::ObjectPtrT<log4cxx::Logger> const&, log4cxx::spi::LocationInfo const&, char const*, ...)':
openrave.cpp:(.text._ZN8OpenRAVE22RavePrintfA_DEBUGLEVELERKN7log4cxx7helpers10ObjectPtrTINS0_6LoggerEEERKNS0_3spi12LocationInfoEPKcz[_ZN8OpenRAVE22RavePrintfA_DEBUGLEVELERKN7log4cxx7helpers10ObjectPtrTINS0_6LoggerEEERKNS0_3spi12LocationInfoEPKcz]+0x28e): undefined reference to `log4cxx::Logger::forcedLog(log4cxx::helpers::ObjectPtrT<log4cxx::Level> const&, std::string const&, log4cxx::spi::LocationInfo const&) const'
libopenrave/libopenrave0.9.so.0.9.0: undefined reference to `log4cxx::Layout::appendFooter(std::string&, log4cxx::helpers::Pool&)'
libopenrave/libopenrave0.9.so.0.9.0: undefined reference to `log4cxx::Layout::getContentType() const'
libopenrave/libopenrave0.9.so.0.9.0: undefined reference to `log4cxx::PatternLayout::PatternLayout(std::string const&)'
libopenrave/libopenrave0.9.so.0.9.0: undefined reference to `log4cxx::Level::Level(int, std::string const&, int)'
libopenrave/libopenrave0.9.so.0.9.0: undefined reference to `log4cxx::Layout::appendHeader(std::string&, log4cxx::helpers::Pool&)'
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:102: recipe for target 'src/openrave0.9' failed

```

I have logc4xx installed (apt-get) on my computer. Does it need a particular version? 
It builds if I disable the OPT_LOG4CXX flag, but then import openravepy exits with a seg fault.
gdb doesn't help and gives me an empty stack.

```

(gdb) r
Starting program: /usr/bin/python -c import\ openravepy
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Inferior 1 (process 22473) exited normally]
(gdb) bt
No stack.

```

sirop, 2017-02-06:
```
sudo apt-get install build-essential cmake doxygen g++ git ipython octave python-dev  python-h5py python-numpy python-pip python-scipy pyt hon-sympy qt4-default minizip

sudo apt-get install ann-tools cmake libann-dev libassimp-dev libavcodec-dev   libavformat-dev libboost-python-dev libboost-all-dev  libccd-dev  libcollada-dom2.4-dp-dev libeigen3-dev libflann-dev liblapack-dev liblog4cxx-dev  libminizip-dev liboctave-dev libode-dev libpcre3-dev libqhu ll-dev libsoqt-dev-common  libsoqt4-dev libswscale-dev libtinyxml-dev libxml2-dev

pip install --upgrade --user sympy==0.7.1

git clone https://github.com/rdiankov/openrave.git -b production --single-branch
cd openrave
mkdir build ; cd build
cmake -DODE_USE_MULTITHREAD=ON -DOPENRAVE_PLUGIN_MOBYRAVE=OFF -DOPT_FCL_COLLISION=OFF -DOPT_QTOSG_VIEWER=OFF -DOPT_DOUBLE_PRECISION=ON ..
make -j `nproc`
```

```
$ dpkg -l | grep log4
ii  liblog4cxx-dev                               0.10.0-10ubuntu1                                            amd64        Logging library for C++ (development files)
ii  liblog4cxx10v5:amd64                         0.10.0-10ubuntu1                                            amd64        Logging library for C++
```

So openrave can be built on Ubuntu 16.04.1 .

crtassos, 2018-09-30:
Nop it doesn't work

sirop, 2018-10-08:
@crtassos 
Please show all of your install steps,
including compiler version and `git status | head-10 `.

skrede, 2018-10-11:
I get the same error as crtassos only for a different target
```
Scanning dependencies of target openrave
[ 49%] Building CXX object src/CMakeFiles/openrave.dir/openrave.cpp.o
[ 49%] Linking CXX executable openrave0.11
libopenrave-core/libopenrave0.11-core.so.0.11.2: undefined reference to OpenRAVE::InterfaceBase::SupportsJSONCommand(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
src/CMakeFiles/openrave.dir/build.make:109: recipe for target 'src/openrave0.11' failed
make[2]: *** [src/openrave0.11] Error 1
CMakeFiles/Makefile2:614: recipe for target 'src/CMakeFiles/openrave.dir/all' failed
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
I followed the steps you wrote [above](https://github.com/rdiankov/openrave/issues/458#issuecomment-277670593) with a slight difference in the cmake invocation
```
cmake -DODE_USE_MULTITHREAD=ON -DOPENRAVE_PLUGIN_MOBYRAVE=OFF -DOPT_FCL_COLLISION=OFF -DOPT_QTOSG_VIEWER=OFF -DOPT_DOUBLE_PRECISION=ON -DCMAKE_INSTALL_PREFIX=/opt/OpenRAVE ..
```

Compiler version:
```
g++ (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

sirop, 2018-10-11:
@skrede 
Well, I used ubuntu 16 and production branch was a bit different those days.
I'll try the same with Ubuntu 18 today and report then.

skrede, 2018-10-11:
Thank you, let me know if there is something else I can do.

sirop, 2018-10-12:
@skrede 

Well, got meanwhile to this https://github.com/rdiankov/openrave/issues/542 .
But there is a solution: https://github.com/rdiankov/openrave/issues/551 
On Ubuntu 18.04 with:
```
~/openrave$ git branch
* production

~/openrave$ git log | head -10
commit 76b31983629d4035269171aadb59fda170fd793f
Author: rosen <rosen.diankov@mujin.co.jp>
Date:   Tue Oct 9 19:45:33 2018 +0900

    change jitter for ikfast
```

sirop, 2018-10-12:
```
 sudo apt-get install cmake g++ git ipython minizip python-dev python-h5py python-numpy python-scipy python-sympy qt4-dev-tools
    2  sudo apt-get install libassimp-dev libavcodec-dev libavformat-dev libavformat-dev libboost-all-dev libboost-date-time-dev libbullet-dev libfaac-dev libglew-dev libgsm1-dev liblapack-dev liblog4cxx-dev libmpfr-dev libode-dev libogg-dev libpcrecpp0v5 libpcre3-dev libqhull-dev libqt4-dev libsoqt-dev-common libsoqt4-dev libswscale-dev libswscale-dev libvorbis-dev libx264-dev libxml2-dev libxvidcore-dev
    3  git clone https://github.com/rdiankov/collada-dom.git
    4  cd collada-dom && mkdir build && cd build
    5  cmake ..
    6  make -j3
    7  sudo make install
    8  cd
    9  pwd
   10  sudo apt-get install libcairo2-dev libjasper-dev libpoppler-glib-dev libsdl2-dev libtiff5-dev libxrandr-dev
   11  sudo apt-get install libcairo2-dev  libpoppler-glib-dev libsdl2-dev libtiff5-dev libxrandr-dev
   22  sudo add-apt-repository "deb http://security.ubuntu.com/ubuntu xenial-security main"
   23  sudo apt update
   24  sudo apt install libjasper1 libjasper-dev
   25  git clone --branch OpenSceneGraph-3.4
   26  git clone --branch OpenSceneGraph-3.4 https://github.com/openscenegraph/OpenSceneGraph.git
   27  cd OpenSceneGraph && mkdir build && cd build
   28  cmake .. -DDESIRED_QT_VERSION=4
   29  make -j3
   30  sudo make install
```

```
  # to uninstall deb version
sudo apt-get -y --purge remove libboost-all-dev libboost-doc libboost-dev
# to uninstall the version which we installed from source
sudo rm -f /usr/lib/libboost_*
 67  sudo apt-get -y install build-essential g++ python-dev autotools-dev libicu-dev libbz2-dev
 68  wget http://downloads.sourceforge.net/project/boost/boost/1.58.0/boost_1_58_0.tar.gz
 69  tar -zxvf boost_1_58_0.tar.gz boost_1_58_0/
tar -zxvf boost_1_58_0.tar.gz
cd boost_1_58_0
# get the no of cpucores to make faster
cpuCores=`cat /proc/cpuinfo | grep "cpu cores" | uniq | awk '{print $NF}'`
echo "Available CPU cores: "$cpuCores
sudo ./b2 --with=all -j $cpuCores install
make -j3
 71  cat /usr/local/include/boost/version.hpp | grep "BOOST_LIB_VERSION"
```

```
 49  sudo apt-get install rapidjson rapidjson-dev
 50  sudo apt-get install  rapidjson-dev
 41  pip install --upgrade --user sympy==0.7.1
 42  sudo apt install python-pip
 43  cd ..
 44  pip install --upgrade --user sympy==0.7.1
 45  git clone https://github.com/rdiankov/openrave.git -b production --single-branch
 46  cd openrave && mkdir build && cd build
 47  ls /usr/local/lib64/
 48  cmake  -DOSG_DIR=/usr/local/lib64/  -DODE_USE_MULTITHREAD=ON -DOPENRAVE_PLUGIN_MOBYRAVE=OFF -DOPT_FCL_COLLISION=OFF -DOPT_QTOSG_VIEWER=ON -DOPT_DOUBLE_PRECISION=ON ..
 make -j3
```
Had some problems with FCL and turned it off therefore. Neither did I yet try out any examples but the compilation works. Do not pay attention to my line numbers.

```
 git branch
* production
cps@ubuntu1:~/openrave/build$ git log | head 5
head: cannot open '5' for reading: No such file or directory
cps@ubuntu1:~/openrave/build$ git log | head -5
commit 76b31983629d4035269171aadb59fda170fd793f
Author: rosen <rosen.diankov@mujin.co.jp>
Date:   Tue Oct 9 19:45:33 2018 +0900

    change jitter for ikfast
```
on Ubuntu 18.04 .

skrede, 2018-10-16:
I'm not so sure adding xenial repositories to bionic is a good idea. This will break the system at one point or another. I appreciate the effort, however

sirop, 2018-10-16:
@skrede 
Guess if you do not need OpenSceneGraph, then  you do not need
```
sudo add-apt-repository "deb http://security.ubuntu.com/ubuntu xenial-security main"
sudo apt update
sudo apt install libjasper1 libjasper-dev
```

PeterBowman, 2018-12-24:
https://github.com/rdiankov/openrave/issues/458#issuecomment-428905260:

> I get the same error as crtassos only for a different target

> libopenrave-core/libopenrave0.11-core.so.0.11.2: undefined reference to OpenRAVE::InterfaceBase::SupportsJSONCommand(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'

Targeted at https://github.com/rdiankov/openrave/pull/629.