rdiankov/openrave #466: C++ in C API

sirop, 2017-01-09:
If you look through https://github.com/rdiankov/openrave/blob/master/src/libopenrave/libopenrave_c.cpp,
you'll find things like default functions values  and bool data type which are valid in C++, but not in C.

I made some cleanup  https://github.com/sirop/openrave/commit/3e4035385a81a92f36fb3860f8a8ab09ca6a64fd .
That's good enough for my needs, as I can use gcc instead of g++ now.

rdiankov, 2017-01-10:
thanks! can you send a pull request?
the C API is only to get C compatible function signatures to do foreign function interchange between various languages, but i have no problem if you want to help get a C compiler to work on it.
best,

sirop, 2017-01-10:
I'll do a proper commit in the next days as https://github.com/sirop/openrave/commit/3e4035385a81a92f36fb3860f8a8ab09ca6a64fd does not contain the C API changes within https://github.com/rdiankov/openrave/tree/master/src/csharpbindings_c .

sirop, 2017-01-13:
Closed via https://github.com/rdiankov/openrave/pull/468 .

closed 2017-01-13 by sirop