rdiankov/openrave #467: Code works in Ubuntu 12.04 but not in Ubuntu 14.04

EdsterG, 2017-01-09:
The following snippet of code works in Ubuntu 12.04 (openrave installed from apt-get)
but not in Ubuntu 14.04 (openrave install from source):
`import openravepy
import numpy as np

env = openravepy.Environment()
env.StopSimulation()
env.Load("robots/pr2-beta-static.zae")
robot = env.GetRobots()[0]
manip = robot.GetManipulator("rightarm")
link = robot.GetLink("r_gripper_tool_frame")

quat_target = [1,0,0,0] # wxyz
xyz_target = [6.51073449e-01,  -1.87673551e-01, 4.91061915e-01]
T_w_link = openravepy.matrixFromPose(quat_target+xyz_target)

Tcur_w_link = link.GetTransform()
Tcur_w_ee = manip.GetEndEffectorTransform()
Tf_link_ee = np.linalg.solve(Tcur_w_link, Tcur_w_ee)

T_w_ee = T_w_link.dot(Tf_link_ee)

print manip.FindIKSolution(T_w_ee, openravepy.IkFilterOptions.CheckEnvCollisions)`

EdsterG, 2017-01-11:
That same snipped of code also don't work in OSX.

sirop, 2017-01-12:
Try to generate solutions first for the appropriate IKType.

EdsterG, 2017-01-13:
The solutions have been generated and are stored in the openrave ik cache. I know this because ikmodel.load() returns True when I run:

```
robot.SetActiveManipulator(manip)
ikmodel = openravepy.databases.inversekinematics.InverseKinematicsModel(
    robot, iktype=openravepy.IkParameterization.Type.Transform6D)
print ikmodel.load()
```

sirop, 2017-01-13:
Just ran this on Ubuntu 14.4
```
import openravepy
import numpy as np

env = openravepy.Environment()
env.StopSimulation()
env.Load("robots/pr2-beta-static.zae")
robot = env.GetRobots()[0]
manip = robot.GetManipulator("rightarm")
link = robot.GetLink("r_gripper_tool_frame")

quat_target = [1,0,0,0] # wxyz
xyz_target = [6.51073449e-01,  -1.87673551e-01, 4.91061915e-01]
T_w_link = openravepy.matrixFromPose(quat_target+xyz_target)

Tcur_w_link = link.GetTransform()
Tcur_w_ee = manip.GetEndEffectorTransform()
Tf_link_ee = np.linalg.solve(Tcur_w_link, Tcur_w_ee)

T_w_ee = T_w_link.dot(Tf_link_ee)

robot.SetActiveManipulator(manip)

from openravepy.misc import *
InitOpenRAVELogging()
ikmodel = openravepy.databases.inversekinematics.InverseKinematicsModel(
    robot, iktype=openravepy.IkParameterization.Type.Transform6D)
ikmodel.autogenerate()
print manip.FindIKSolution(T_w_ee, openravepy.IkFilterOptions.CheckEnvCollisions)
```

Got as solution:
```
[  6.26493171e-04   1.29563325e+00   0.00000000e+00  -1.52774063e+00
   3.13886912e+00  -2.32108207e-01  -3.13894215e+00]
```

EdsterG, 2017-01-14:
I probably should have specified that I used the same hack/workaround before I opened this issue.
Why doesn't openrave autoload the models in 14.04 like it does in 12.04? There should be no need to run the following code for every manipulator:

```
robot.SetActiveManipulator(manip)
ikmodel = openravepy.databases.inversekinematics.InverseKinematicsModel(
    robot, iktype=openravepy.IkParameterization.Type.Transform6D)
if not ikmodel.load():
    ikmodel.autogenerate()
```

sirop, 2017-01-14:
```
if not ikmodel.load():
    ikmodel.autogenerate()
```
works  at my machine.

I used the master branch for building from sources.

I

rdiankov, 2017-01-14:
to tell you the truth, i am not sure how IK was loading in the initial snippet .... the correct thing is to use ikmodel.load() like shown later.

closed 2017-01-14 by rdiankov