rdiankov/openrave #469: Different joint limits in Ubuntu 12.04 vs Ubuntu 14.04

EdsterG, 2017-01-14:
The following snippet of code returns different joint limits in Ubuntu 12.04 (openrave installed from apt-get) and Ubuntu 14.04 (openrave install from source):

```
import openravepy

env = openravepy.Environment()
env.StopSimulation()
env.Load("robots/pr2-beta-static.zae")
robot = env.GetRobots()[0]

active_dofs = robot.GetManipulator('leftarm').GetArmIndices()
robot.SetActiveDOFs(active_dofs)
print robot.GetActiveDOFLimits()
```

In 14.04:
```
Lower:
-0.564601795715
-0.353600215796
-0.650000756015
-2.12130807946
-10000.0
-2.00000769645
-10000.0
Upper:
2.13539288652
1.29629966869
3.74999697754
-0.150000053635
10000.0
-0.100000035756
10000.0
```

In 12.04:
```
Lower:
-0.714602372949
-0.523600520928
-0.79999958792
-2.32130535844
-10000.0
-2.1800035022
-10000.0
Upper:
2.28540393573
1.39630005351
3.90000802675
0.0
10000.0
0.0
10000.0
```

Why are there such large variations in the lower joint limits?

sirop, 2017-01-14:
Get these warnings too when loading `pr2-beta-static.zae` on Ubuntu 14.04 (master branch)
```
2017-01-14 02:43:22,432 openrave [WARN] [kinbody.cpp:1524 KinBody::SetDOFValues] dof 12 value is not in limits 0.000000e+00<1.150000e-02
2017-01-14 02:43:22,433 openrave [WARN] [kinbody.cpp:1535 KinBody::SetDOFValues] dof 18 value is not in limits 0.000000e+00<-1.500001e-01
2017-01-14 02:43:22,433 openrave [WARN] [kinbody.cpp:1535 KinBody::SetDOFValues] dof 20 value is not in limits 0.000000e+00<-1.000000e-01
2017-01-14 02:43:22,433 openrave [WARN] [kinbody.cpp:1535 KinBody::SetDOFValues] dof 30 value is not in limits 0.000000e+00<-1.500001e-01
2017-01-14 02:43:22,433 openrave [WARN] [kinbody.cpp:1535 KinBody::SetDOFValues] dof 32 value is not in limits 0.000000e+00<-1.000000e-01
```

The joint position limits of my own models are parsed correctly.

rdiankov, 2017-01-14:
robots/pr2-beta-static.zae might be different on both of your systems, are you sure you are using the same version of openrave for both of them?

looking at the pr2-beta-static.zae, for the first arm dof the correct limits should be [-0.5646018, 2.13539289]

sirop, 2017-01-14:
> looking at the pr2-beta-static.zae, for the first arm dof the correct limits should be [-0.5646018, 2.13539289]  

I get also  [-0.5646018, 2.13539289] on Ubuntu 14.04 with master branch.  

But what about the warnings when loading pr2-beta-static.zae :
`2017-01-14 02:43:22,433 openrave [WARN] [kinbody.cpp:1535 KinBody::SetDOFValues] dof 32 value is not in limits 0.000000e+00<-1.000000e-01`

EdsterG, 2017-01-14:
Ah, yes the models are indeed different. Should I assume the new model has more accurate joint limits?

As for the warnings, these warnings don't show up with the old model because 0 is in the range of all the joint limits. With the new model, certain joints limit don't include 0 in their range.

rdiankov, 2017-01-15:
good to know. i think the warnings are ok, it's just not a good way of modeling the robot..

closed 2017-01-15 by rdiankov

EdsterG, 2017-01-15:
Why did the OpenRave model change between versions?

EdsterG, 2017-01-15:
The current model has incorrect joint limits. The old model had correct joint limits as specified here: https://www.clearpathrobotics.com/wp-content/uploads/2014/08/pr2_manual_r321.pdf

Should probably revert to the old model?