rdiankov/openrave #470: ikfast error : tj13 was not declared

k-okada, 2017-01-18:
Hi, I have following error when compile ikfast generated cpp code

```
In file included from /home/k-okada/catkin_ws/ws_nextage/src/rtmros_nextage/nextage_ik_plugin/src/nextage_left_arm_ikfast_moveit_plugin.cpp:114:0:
/home/k-okada/catkin_ws/ws_nextage/src/rtmros_nextage/nextage_ik_plugin/src/nextage_left_arm_ikfast_solver_latest.cpp:10149:48: error: ‘tj13’ was not declared in this scope
 CheckValue<IkReal> x2917=IKPowWithIntegerCheck(tj13,-1);                                               ^
```

the solver.cpp is generated by lastest ikfast.py
```
 `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot=main.dae --iktype=transform6d --baselink=1 --eelink=15 --savefile=nextage_left_arm_ikfast_solver.cpp
```
you can get main.dae from https://github.com/tork-a/rtmros_nextage/raw/indigo-devel/nextage_description/models/main.dae
and it was working with solve.cpp generated a few years ago (`IKFAST_COMPILE_ASSERT(IKFAST_VERSION==61);`)

it seems something wrong around https://github.com/rdiankov/openrave/blob/master/python/ikfast_generator_cpp.py#L2062-L2065
```
                elif expr.exp.is_integer:
                    # use IKPowWithIntegerCheck in order to make it robust
                    ikpowsymbol = self.symbolgen.next()
                    code2 = cStringIO.StringIO()
                    code2.write('CheckValue<IkReal> %s=IKPowWithIntegerCheck('%ikpowsymbol)
```

Did you forget to instantiate `ikpowsymbol` ?

rdiankov, 2017-01-24:
kei,

thanks for the report!
i just tested on openrave's production (and master) branch version of ikfast and it works without a problem. please try to update the ikfast version from latest openrave.

closed 2017-01-24 by rdiankov

k-okada, 2017-02-02:
rosen
thanks for info, I tried 
18521d4 `Merge branch 'production' of ssh://github.com/rdiankov/openrave into production`
but we still have `‘tj13’ was not declared` problem.
here is whole output -> https://gist.github.com/k-okada/9fbdb0bdad88ba9a4278cf2dff2b5dd3 so if you have noticed something please let me know. or could you share output of your environment? so I can try to figure out the difference.

liuhuanjim013, 2017-02-02:
hi kei,

it also worked for me with this hash: f138ae0e5bc54f2537ff6aac4dcdc0345f8e3efd . 

maybe i can take a look at your openrave cmake log. it's possible that one of the libraries openrave depends on is old.

k-okada, 2017-02-02:
@liuhuanjim013 thanks for your kind support! here is my cmake output.
```
[openrave:make] cd /home/k-okada/catkin_ws/ws_openrave/build/openrave/build_openrave && cmake /home/k-okada/catkin_ws/ws_openrave/build/openrave/source_openrave -DOPT_FCL_COLLISION=OFF -DCMAKE_INSTALL_PREFIX=/home/k-okada/catkin_ws/ws_openrave/devel/.private/openrave
[openrave:make] Re-run cmake no build system arguments                                                                                                                           
[openrave:make] -- Compiling OpenRAVE Version 0.9.0, soversion=0.9                                                                                                               
[openrave:make] -- Using cmake version 2.8.12                                                                                                                                    
[openrave:make] -- Using double precision                                                                                                                                        
[openrave:make] -- detected system processor: x86_64                                                                                                                             
[openrave:make] -- installing to /home/k-okada/catkin_ws/ws_openrave/devel/.private/openrave, python exe /usr/bin/python, python subdir is lib/python2.7/dist-packages, program suffix is '0.9'
[openrave:make] -- Boost version: 1.54.0                                                                                                                                         
[openrave:make] -- Found the following Boost libraries:                                                                                                                          
[openrave:make] --   regex                                                                                                                                                       
[openrave:make] --   filesystem                                                                                                                                                  
[openrave:make] --   system                                                                                                                                                      
[openrave:make] --   python                                                                                                                                                      
[openrave:make] --   thread                                                                                                                                                      
[openrave:make] --   iostreams                                                                                                                                                   
[openrave:make] --   date_time                                                                                                                                                   
[openrave:make] -- found boost version: 105400                                                                                                                                   
[openrave:make] -- Targetting Unix Makefiles                                                                                                                                     
[openrave:make] -- Using /usr/bin/c++ compiler.                                                                                                                                  
[openrave:make] -- Could NOT find MPFI                                                                                                                                           
[openrave:make] -- Using local crlibm                                                                                                                                            
[openrave:make] -- compiling with double extended (long double) floating point                                                                                                   
[openrave:make] CMake Warning at CMakeLists.txt:480 (find_package):                                                                                                              
[openrave:make]   By not providing "FindASSIMP.cmake" in CMAKE_MODULE_PATH this project has                                                                                      
[openrave:make]   asked CMake to find a package configuration file provided by "ASSIMP", but                                                                                     
[openrave:make]   CMake did not find one.                                                                                                                                        
[openrave:make]                                                                                                                                                                  
[openrave:make]   Could not find a package configuration file provided by "ASSIMP" with any                                                                                      
[openrave:make]   of the following names:                                                                                                                                        
[... cut ...]
[openrave:make] -- ODE found, building oderave plugin, precision=DOUBLE                                                                                                          
[openrave:make] -- Configuring done                                                                                                                                              
[openrave:make] -- Generating done                                                                                                                                               
[openrave:make] -- Build files have been written to: /home/k-okada/catkin_ws/ws_openrave/build/openrave/build_openrave    
```

liuhuanjim013, 2017-02-02:
one thing i noticed is sympy version. can you try 0.7.1?

k-okada, 2017-02-02:
@liuhuanjim013 great! it works!! thank  you very much.
by the why, how did you install sympy? I'm using ubuntu 14.04 and the default sympy is 0.7.4 and I have to install via pip.

liuhuanjim013, 2017-02-02:
Awesome! Yep, pip is the way to go :)

k-okada, 2017-02-02:
I see, anyway thank you very much again!!!

MuneebUlHaq, 2017-05-05:
Please let me know , how one can roll back from sympy 1.0 to sympy 0.7.1.. Thanks

ompugao, 2017-05-05:
pip install sympy==0.7.1

MuneebUlHaq, 2017-05-05:
Thanks for the answer !
I tried but i am getting this error:

muneebul@ubuntu:~$ pip install sympy==0.7.1
Downloading/unpacking sympy==0.7.1
  Downloading sympy-0.7.1.tar.gz (3.6MB): 3.6MB downloaded
  Running setup.py (path:/tmp/pip_build_muneebul/sympy/setup.py) egg_info
for package sympy

    package init file 'sympy/assumptions/tests/__init__.py' not found (or
not a regular file)
    package init file 'sympy/concrete/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/core/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/external/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/functions/combinatorial/tests/__init__.py' not
found (or not a regular file)
    package init file 'sympy/functions/elementary/tests/__init__.py' not
found (or not a regular file)
    package init file 'sympy/functions/special/tests/__init__.py' not found
(or not a regular file)
    package init file 'sympy/galgebra/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/geometry/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/integrals/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/logic/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/matrices/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/ntheory/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/parsing/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/physics/quantum/tests/__init__.py' not found
(or not a regular file)
    package init file 'sympy/physics/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/plotting/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/polys/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/printing/pretty/tests/__init__.py' not found
(or not a regular file)
    package init file 'sympy/printing/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/series/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/simplify/tests/__init__.py' not found (or not
a regular file)
    package init file 'sympy/slow_tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/solvers/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/statistics/tests/__init__.py' not found (or
not a regular file)
    package init file 'sympy/tensor/tests/__init__.py' not found (or not a
regular file)
    package init file 'sympy/utilities/tests/__init__.py' not found (or not
a regular file)
Installing collected packages: sympy
  Found existing installation: sympy 1.0
    Uninstalling sympy:
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in
main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line
283, in run
    requirement_set.install(install_options, global_options,
root=options.root_path)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1432, in install
    requirement.uninstall(auto_confirm=True)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 599, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1845, in remove
    renames(path, new_path)
  File "/usr/lib/python2.7/dist-packages/pip/util.py", line 296, in renames
    shutil.move(old, new)
  File "/usr/lib/python2.7/shutil.py", line 303, in move
    os.unlink(src)
OSError: [Errno 13] Permission denied: '/usr/local/bin/isympy'

Storing debug log for failure in /home/muneebul/.pip/pip.log

Puttichai, 2017-05-05:
@MuneebUlHaq `sudo pip install sympy==0.7.1`