rdiankov/openrave #479: Make Error: ‘const class std::vector<int>’ has no member named ‘find’

fredsukkar, 2017-04-04:
Issue appears to stem from commit: `fb078fc0848788218ba1ac225fe176c3c07fe204`

Changing adjacent links from sets to vectors throws an error because `std::vector` has no member `find`.

rdiankov, 2017-04-05:
oops, for some reason it is ok on gcc 4.7... what gcc version are you using?
can you give me the error log? it'll easier to fix all the places this way.

thanks

fredsukkar, 2017-04-10:
Error log:
`In file included from /home/fred/git/openrave/plugins/bulletrave/bulletrave.cpp:18:0:
/home/fred/git/openrave/plugins/bulletrave/bulletcollision.h: In member function ‘virtual bool BulletCollisionChecker::LinkAdjacentFilterCallback::CheckLinks(OpenRAVE::KinBody::LinkPtr, OpenRAVE::KinBody::LinkPtr) const’:`
`/home/fred/git/openrave/plugins/bulletrave/bulletcollision.h:171:34: error: ‘const class std::vector<int>’ has no member named ‘find’`
  `           return _setadjacency.find(index0|(index1<<16)) != _setadjacency.end() || _setadjacency.find(index1|(index0<<16)) != _setadjacency.end();`
                                  ^
`/home/fred/git/openrave/plugins/bulletrave/bulletcollision.h:171:100: error: ‘const class std::vector<int>’ has no member named ‘find’`
`             return _setadjacency.find(index0|(index1<<16)) != _setadjacency.end() || _setadjacency.find(index1|(index0<<16)) != _setadjacency.end();`
`/home/fred/git/openrave/plugins/bulletrave/bulletcollision.h:172:9: warning: control reaches end of non-void function [-Wreturn-type]`
         }
         ^
`make[2]: *** [plugins/bulletrave/CMakeFiles/bulletrave.dir/bulletrave.cpp.o] Error 1`
`make[1]: *** [plugins/bulletrave/CMakeFiles/bulletrave.dir/all] Error 2`

hamzamerzic, 2017-04-26:
Have the same error with gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4.

rdiankov, 2017-08-12:
should be fixed now.

closed 2017-08-12 by rdiankov