rdiankov/openrave #483: Failed to compile under ubuntu 16.04

Gummary, 2017-04-11:
Hello,
I am trying to compile the latest_stable branch on16.04. After running cmake, make is failing with the following error:

`3rdparty/fparser-4.5/CMakeFiles/fparser.dir/build.make:62: recipe for target '3rdparty/fparser-4.5/CMakeFiles/fparser.dir/fparser.cc.o' failed
make[2]: *** [3rdparty/fparser-4.5/CMakeFiles/fparser.dir/fparser.cc.o] Error 1
CMakeFiles/Makefile2:546: recipe for target '3rdparty/fparser-4.5/CMakeFiles/fparser.dir/all' failed
make[1]: *** [3rdparty/fparser-4.5/CMakeFiles/fparser.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2`

My steps is follow [this artical](https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html)

fuzzyTew, 2017-06-30:
On debian stretch, using GCC 6.3.0, I've found I had to update the local fparser library from 4.5 to 4.5.2 to get make to continue past it.  This library is available at http://warp.povusers.org/FunctionParser/ and should be unpacked in `3rdparty/fparser-4.5`

disdaining, 2017-08-31:
After extracting contents from 4.5.2 to 4.5 under linux mint 18.2, ccmake gives the following error:
```
CMake Warning at CMakeLists.txt:710 (find_package):
   By not providing "FindFPARSER.cmake" in CMAKE_MODULE_PATH this project has
   asked CMake to find a package configuration file provided by "FPARSER", but
   CMake did not find one.

   Could not find a package configuration file provided by "FPARSER"
   (requested version 4.4.3) with any of the following names:

     FPARSERConfig.cmake
     fparser-config.cmake

   Add the installation prefix of "FPARSER" to CMAKE_PREFIX_PATH or set
   "FPARSER_DIR" to a directory containing one of the above files.  If
   "FPARSER" provides a separate development package or SDK, be sure it has
   been installed.

 CMake Error at CMakeLists.txt:811 (install):
   install DIRECTORY given no DESTINATION!

 CMake Error at python/CMakeLists.txt:7 (install):

```