rdiankov/openrave #491: openrave installed in windows 10?

Joylim, 2017-06-07:
Hi, anyone uses openrave in windows? Need your help!!

imcardle, 2019-02-04:
I can't seem to get the windows installer (openrave-0.9.0-5cfc74-win32-vc100-setup.exe) to work for me on my Windows 10.  It pops up an error during installation of the boost 1.44 library (see attached image).  Is this a known issue?  Any workaround?
[image]

johannnovak, 2021-05-22:
Hi,
I got the same issue. Still no workaround ?

Thanks

cielavenir, 2021-05-23:
Where can I find the installer?

Anyway you should download boost from https://sourceforge.net/projects/boost/files/boost-binaries/

Although there are no results running openrave on Windows (except WSL)...

cielavenir, 2022-05-01:
(that movie seems to use Linux (apperantly GNOME), not Windows)

guiglass, 2023-05-28:
Same issue