rdiankov/openrave #494: Ubuntu 16.04 Source Compile server not functioning

jerinphilip, 2017-06-12:
Hello, 

I'm trying to get openrave latest to run on my Ubuntu 16.04, and later run experiments on Matlab. This requires the server be running on port 4765 (default port). 

However, when I try to run my matlab code, I'm getting errors on OrCommunicator. When I check

`netstat -tlnp`

I find nothing running on port 4765. I can get more debug/logs if required. Openrave viewer and all is loading, just no server. I already tried increasing the debug level, came up with nothing.

How can I fix this?

jerinphilip, 2017-06-12:
https://github.com/rdiankov/openrave/blob/master/src/openrave.cpp#L83

This is causing the error. Why is "textserver" commented out?

rdiankov, 2017-08-24:
try starting with "-server" option, which would create the text server

closed 2017-08-24 by rdiankov