rdiankov/openrave #495: GUI segfaults after pressing the cube button on the right hand side

dzhao14, 2017-06-12:
I've installed openrave following this [link](https://scaron.info/teaching/installing-openrave-on-ubuntu-14.04.html). I get an issue when I run `openrave.py --example graspplanning` First off if I just watch the program it cannot grasp the mugs since in the termial I get this output:
`2017-06-12 11:41:09,898 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 1
openravepy.databases.inversekinematics: GetDefaultIndices, found 3-intersection centered on index 5
robot resolutions:  [ 0.00396017  0.00455076  0.00593835  0.00951787  0.01218716  0.0175349
  0.02203869  0.0606854   0.0606854   0.0606854   0.04497103]
robot weights:  [ 2.77253268  2.41271402  1.84894655  1.15358696  0.90092279  0.62616198
  0.49820064  0.18092804  0.18092804  0.18092804  0.24415027]
searching for graspable objects (robot=42daa21cdb5bcad0df5d94673fd5d61a)...
mug1 is graspable
mug2 is graspable
mug3 is graspable
mug4 is graspable
mug5 is graspable
mug6 is graspable
searching for destinations on table...
starting to pick and place random objects
grasping object mug3
2017-06-12 11:41:11,695 openrave [ERROR] [taskmanipulation.cpp:517 GraspPlanning] failed processing command igrasptrans
failed to grasp object mug3
Planning Error

grasping object mug1
2017-06-12 11:41:11,713 openrave [ERROR] [taskmanipulation.cpp:517 GraspPlanning] failed processing command igrasptrans
failed to grasp object mug1
Planning Error
`

The bigger issue though, is that whenever I press the red cube button on the right hand side three times the program crashes with a segfault.

Is there something wrong that I did during the install process? When I ran the test using `python run_tests.py` It seemed like the graspplanning worked there but some of the tests there failed as well.
What should I do to remedy this issue?

JimmyLaw007, 2018-05-18:
**I also encountered such a problem. 
Will you solve it? 
Can you share your solution?
Thank you！**

lakshmip001, 2019-03-13:
Do you have any solution for solving this error.