rdiankov/openrave #498: Ubuntu 16.04 make compile failure

NotCras, 2017-06-22:
Can't build openrave on my system. 
Xubuntu 16.04, Kernel 4.4.0-81-generic, cmake version 3.5.1

the cmake command settings I've tried:
`cmake -DODE_USE_MULTITHREAD=ON -DOPENRAVE_PLUGIN_MOBYRAVE=OFF -DOPT_FCL_COLLISION=OFF -DOPT_QTOSG_VIEWER=OFF -DOPT_DOUBLE_PRECISION=ON  -DOSG_DIR=/usr/local/lib64/ ..`

and 

`cmake -DODE_USE_MULTITHREAD=ON -DOSG_DIR=/usr/local/lib64/ ..`

Since there are a ton of deprecated function warnings and notes that come up with a normal make command, I show the output of:

`make -wno-deprecated-declarations -j 8`

This make command (using the latter cmake command) produces the following errors:

`
make[2]: *** No rule to make target 'plugins/rplanners/ParabolicPathSmooth/libParabolicPathSmooth.a', needed by 'plugins/rplanners/ParabolicPathSmooth/testparabolicramp'.  Stop.
...
make[1]: *** [plugins/rplanners/ParabolicPathSmooth/CMakeFiles/testparabolicramp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
...
make[2]: *** No rule to make target 'plugins/rplanners/ParabolicPathSmooth/libParabolicPathSmooth.a', needed by 'plugins/rplanners/librplanners.so'.  Stop.
...
make[1]: *** [plugins/rplanners/CMakeFiles/rplanners.dir/all] Error 2
...
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
make: Leaving directory '/home/cras/Programs/openrave/build'
`
The ParabolicPathSmooth plugin does exist. I don't know how to check if things have been linked wrong.

In previous make attempts its also thrown errors for other components, but its gotten stuck on this. Currently following two installation walkthroughs: [here](https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html) and [here](https://fsuarez6.github.io/blog/workstation-setup-xenial/).

NotCras, 2017-06-23:
Also, to add, here is my cmake output for further debugging:

    -- Compiling OpenRAVE Version 0.9.0, soversion=0.9
    -- Using cmake version 3.5.1
    -- Using double precision
    -- detected system processor: x86_64
    -- installing to /usr/local, python exe /usr/bin/python, python subdir is lib/python2.7/dist-packages, program suffix is '0.9'
    -- Boost version: 1.58.0
    -- Found the following Boost libraries:
    --   regex
    --   filesystem
    --   system
    --   python
    --   thread
    --   iostreams
    --   date_time
    --   chrono
    --   atomic
    -- found boost version: 105800
    -- Targetting Unix Makefiles
    -- Using /usr/bin/c++ compiler.
    -- Could NOT find MPFI
    -- Using local crlibm
    -- compiling with double extended (long double) floating point
    -- assimp version 3.2.0, , /usr/lib/include
    CMake Warning (dev) at modules-cmake/CheckCXXSourceRuns.cmake:16 (IF):
      Policy CMP0054 is not set: Only interpret if() arguments as variables or
      keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
      details.  Use the cmake_policy command to set the policy and suppress this
      warning.

      Quoted variables like "IS_X86_64" will no longer be dereferenced when the
      policy is set to NEW.  Since the policy is not set the OLD behavior will be
      used.
    Call Stack (most recent call first):
      CMakeLists.txt:505 (check_cxx_source_runs)
    This warning is for project developers.  Use -Wno-dev to suppress it.

    -- WARNING: you are using the obsolete 'PKGCONFIG' macro, use FindPkgConfig
    -- libxml2 found
    -- Checking for module 'minizip'
    --   No package 'minizip' found
    -- compiling local ann library
    CMake Warning at CMakeLists.txt:710 (find_package):
      By not providing "FindFPARSER.cmake" in CMAKE_MODULE_PATH this project has
      asked CMake to find a package configuration file provided by "FPARSER", but
      CMake did not find one.

      Could not find a package configuration file provided by "FPARSER"
      (requested version 4.4.3) with any of the following names:

        FPARSERConfig.cmake
        fparser-config.cmake

      Add the installation prefix of "FPARSER" to CMAKE_PREFIX_PATH or set
      "FPARSER_DIR" to a directory containing one of the above files.  If
      "FPARSER" provides a separate development package or SDK, be sure it has
      been installed.

    -- Checking for one of the modules 'fparser'
    -- compiling local fparser library
    -- compiling local convexdecomposition library
    -- Geometry Parsing: Collada DOM 2.5.0
    -- Geometry Parsing: Using Open Asset Import Library 3.2.0
    -- Geometry Parsing: Using ivcon
    -- extracting models to /home/cras/Programs/openrave/src
    -- MATLAB installation not found, is mex in the system path?
    -- Octave installation not found, is 'mkoctfile' in the system path?
    -- Found sympy version '0.7.1'
    -- Assuming sympy version 0.7.x or greater
    CMake Warning (dev) at modules-cmake/CheckCXXSourceRuns.cmake:16 (IF):
      Policy CMP0054 is not set: Only interpret if() arguments as variables or
      keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
      details.  Use the cmake_policy command to set the policy and suppress this
      warning.

      Quoted variables like "HAVE_ALL_PYTHON_HEADERS" will no longer be
      dereferenced when the policy is set to NEW.  Since the policy is not set
      the OLD behavior will be used.
    Call Stack (most recent call first):
      python/bindings/CMakeLists.txt:13 (check_cxx_source_runs)
    This warning is for project developers.  Use -Wno-dev to suppress it.

    -- python and boost-python found
    -- unsupported bullet version 2.83
    -- FindMoby: Could not find EventDrivenSimulator.h
    -- Could not find Moby Physics engine. Please install from https://github.com/PositronicsLab/Moby 
    -- enabling FFMPEG support
    -- boost regex found
    -- Found osgDB: /usr/local/lib64/libosgDB.so  
    -- Found osgQt: /usr/local/lib64/libosgQt.so  
    -- Found osgGA: /usr/local/lib64/libosgGA.so  
    -- Found osgFX: /usr/local/lib64/libosgFX.so  
    -- Found osgText: /usr/local/lib64/libosgText.so  
    -- Found osgViewer: /usr/local/lib64/libosgViewer.so  
    -- Found osgManipulator: /usr/local/lib64/libosgManipulator.so  
    -- Found osg: /usr/local/lib64/libosg.so  
    -- Found OpenThreads: /usr/local/lib64/libOpenThreads.so  
    -- Found OpenSceneGraph:         /usr/local/lib64/libosgDB.so;/usr/local/lib64/libosgQt.so;/usr/local/lib64/libosgGA.so;/usr/local/lib64/libosgFX.so;/usr/local/lib64/libosgText.so;/usr/local/lib64/libosgViewer.so;/usr/local/lib64/libosgManipulator.so;/usr/local/lib64/libosg.so;/usr/local/lib64/libOpenThreads.so (found suitable version "3.4.0", minimum required is "3.4") 
    -- Detected Qt/OSG GUI, making plugin
    -- QT dirs: /usr/include/qt4 OPENGL: /usr/include/qt4/QtOpenGL
    -- Link libraries: /usr/lib/x86_64-linux-gnu/libQtDeclarative.so;/usr/lib/x86_64-linux-gnu/libQtGui.so;/usr/lib/x86_64-linux-gnu/libQtCore.so;/usr/lib/x86_64-linux-gnu/libxml2.so;/usr/lib/x86_64-linux-gnu/libQtOpenGL.so;/usr/lib/x86_64-linux-gnu/libGLU.so;/usr/lib/x86_64-linux-gnu/libGL.so;/usr/local/lib64/libosgDB.so;/usr/local/lib64/libosgQt.so;/usr/local/lib64/libosgGA.so;/usr/local/lib64/libosgFX.so;/usr/local/lib64/libosgText.so;/usr/local/lib64/libosgViewer.so;/usr/local/lib64/libosgManipulator.so;/usr/local/lib64/libosg.so;/usr/local/lib64/libOpenThreads.so
    -- Detected SoQt/Coin3D GUI, making plugin
    -- Detected ODE multi-threading extensions, but not sure if should enable them... define ODE_USE_MULTITHREAD variable to use them
    -- ODE found, building oderave plugin, precision=SINGLE
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/cras/Programs/openrave/build

aydal, 2018-05-28:
Hi,
Did you resolve the issue?

NotCras, 2018-06-12:
No, I didn't. However I think its something with the xubuntu core systems because when I switched to ubuntu it ran installed fine.

closed 2018-06-12 by NotCras