rdiankov/openrave #500: Failed to Install OpenRave on 16.04 Using Instructions

scchow, 2017-06-27:
Hi,

I have been trying to install OpenRave from souce using the instructions found on the main OpenRave [website](http://openrave.org/docs/latest_stable/coreapihtml/installation_linux.html).

The instructions state to run:
```
sudo add-apt-repository ppa:openrave/release
sudo sh -c 'echo "deb-src http://ppa.launchpad.net/openrave/release/ubuntu `lsb_release -cs` main" >> /etc/apt/sources.list.d/openrave-release-`lsb_release -cs`.list'
sudo apt-get update
sudo apt-get build-dep openrave
```

But I get the following error after running `sudo apt-get update `:
```
Reading package lists... Done
W: The repository 'http://ppa.launchpad.net/openrave/release/ubuntu xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/openrave/release/ubuntu/dists/xenial/main/source/Sources  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

It seems like that ppa does not have a Release file for 16.04. I encountered the same issue when trying to install the binary files of OpenRave via:
```
sudo add-apt-repository ppa:openrave/release
sudo apt-get update
sudo apt-get install openrave
```
as described on the [main website](http://openrave.org/docs/latest_stable/install/#install) as well.

scchow, 2017-06-27:
Just an update, I decided to follow the instructions [here](https://scaron.info/teaching/troubleshooting-openrave-installation.html) for installation.

OpenRAVE seemed to install, but I am missing some dependencies. When I run OpenRAVE on my `dae` file, I get the following

```
openrave robot.dae
2017-06-27 15:54:35,049 openrave [WARN] [openrave.cpp:322 MainOpenRAVEThread] failed to find an OpenRAVE viewer.
```
Does anyone know what viewer OpenRAVE is trying to use and where to get it?

maxgittelman, 2017-11-16:
@scchow I have the same issue, did you manage to find the solution?

maxgittelman, 2017-11-16:
Solved, simply install this package through apt `libsoqt4-dev` and source install via [troubleshooting page](https://scaron.info/teaching/troubleshooting-openrave-installation.html)

yijiangh, 2017-12-09:
@maxgittelman Please check the message when you cmake openrave. You might find the following warnings:

```
-- Could not find OpenSceneGraph v3+. Please install OSG (http://www.openscenegraph.org/projects/osg)
-- Detected SoQt/Coin3D GUI, making plugin
WARNING: Could not find ODE. Please install ODE (http://www.ode.org)
```

In my case (Ubuntu 16.04), installing these missing packages:

```
sudo apt-get install libqt4-dev libsoqt-dev-common libsoqt4-dev
```

will fix the problem.

maxgittelman, 2017-12-11:
The one liner didn't do it for me, but by installing all the dependencies found in this [tutorial](https://scaron.info/teaching/installing-openrave-on-ubuntu-16.04.html) I managed to install OpenRAVE.