rdiankov/openrave #520: update website?

ahundt, 2017-08-11:
http://openrave.org/ says `Latest Official Release: 0.8.2`, but the [github releases tab](https://github.com/rdiankov/openrave/releases) says `0.9`. Could this be fixed?

rdiankov, 2017-08-12:
Hi Andrew,

Thanks for the ping.
Unfortunately not all openrave tests are passing with 0.9, so there hasn't been an "official update" yet. However, I highly recommend you use 0.9 since there's been a lot of bug fixes and most likely you'll never run into the parts of the code where tests are not passing.

Cheers,

rdiankov, 2017-08-12:
By the way, if you can help hunt down the rest of the tests that are failing (look in *test* dir), then they'll help speed up putting a new official version out there.

closed 2017-08-12 by rdiankov