rdiankov/openrave #524: OpenRAVE Packaging (Ubuntu PPA)

fsuarez6, 2017-09-06:
Hi guys,
Are there any plans to release the current (or any) version of OpenRAVE to the [Ubuntu PPA](https://launchpad.net/~openrave)?
It would be a really nice to be able to install OpenRAVE using the Ubuntu Package Manager, e.g:
```bash
sudo add-apt-repository ppa:openrave/release
sudo apt-get install openrave
```
Moreover, it will avoid issues like these:
- #346
- #483
- #498
- #516

ardabbour, 2017-11-01:
This is already there, but the latest Ubuntu version it works with is Wily (15.10); see [here](https://launchpad.net/~openrave/+archive/ubuntu/release). It could be updated.

fsuarez6, 2017-11-02:
The PPA exists but OpenRAVE has not been release since Precise (Ubuntu 12.04). That means it's not available for distributions above Precise.
You can check this if you [filter by distribution](https://launchpad.net/~openrave/+archive/ubuntu/release?field.series_filter=wily)