rdiankov/openrave #532: build error: no known conversion from ‘VectorWrapper<const double>’ to ‘const std::vector<double>&’

morxa, 2017-10-06:
The current master (`108a9d812b1a32fa657cf336b3f8dd298803c497`) fails to build on Fedora 26 with the following error (subsequent run to only have relevant messages):

```
[  0%] Built target openrave-md5-native
[  2%] Built target cpp-gen-md5-native
[  3%] Built target interfacehashes_target
[  4%] Built target openrave-md5
[  5%] Built target ivcon
[ 10%] Built target qhull
[ 14%] Built target convexdecomposition
[ 15%] Built target check_libm_accuracy-native
[ 24%] Built target libopenrave
[ 29%] Built target libopenrave-core
[ 30%] Built target openrave
[ 31%] Built target libopenrave_c
[ 32%] Built target libopenrave-core_c
[ 32%] Built target orcreate_octave
[ 32%] Built target orread_octave
[ 33%] Built target orwrite_octave
[ 38%] Built target locale
[ 39%] Built target pyANN_int
[ 40%] Built target convexdecompositionpy
[ 49%] Built target openravepy_int
[ 51%] Built target basecontrollers
[ 53%] Built target baserobots
[ 54%] Built target basesamplers
[ 55%] Built target basesensors
[ 57%] Built target configurationcache
[ 58%] Built target openravepy_configurationcache
[ 59%] Built target dualmanipulation
[ 61%] Built target grasper
[ 67%] Built target ikfastsolvers
[ 68%] Built target logging
[ 70%] Built target PQP
[ 71%] Built target pqprave
[ 74%] Built target rmanipulation
[ 77%] Built target ParabolicPathSmooth
[ 79%] Built target rampoptimizer
[ 79%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/rplanners.cpp.o
In file included from /usr/include/c++/7/x86_64-redhat-linux/bits/c++allocator.h:33:0,
                 from /usr/include/c++/7/bits/allocator.h:46,
                 from /usr/include/c++/7/string:41,
                 from /home/thofmann/code/openrave/include/openrave/openrave.h:55,
                 from /home/thofmann/code/openrave/plugins/include/openraveplugindefs.h:20,
                 from /home/thofmann/code/openrave/plugins/rplanners/rplanners.cpp:17:
/usr/include/c++/7/ext/new_allocator.h: In instantiation of ‘class __gnu_cxx::new_allocator<const double>’:
/usr/include/c++/7/bits/allocator.h:108:11:   required from ‘class std::allocator<const double>’
/usr/include/c++/7/bits/stl_vector.h:81:14:   required from ‘struct std::_Vector_base<const double, std::allocator<const double> >::_Vector_impl’
/usr/include/c++/7/bits/stl_vector.h:166:20:   required from ‘struct std::_Vector_base<const double, std::allocator<const double> >’
/usr/include/c++/7/bits/stl_vector.h:216:11:   required from ‘class std::vector<const double, std::allocator<const double> >’
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h:37:7:   required from ‘class VectorWrapper<const double>’
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h:217:124:   required from here
/usr/include/c++/7/ext/new_allocator.h:93:7: error: ‘const _Tp* __gnu_cxx::new_allocator<_Tp>::address(__gnu_cxx::new_allocator<_Tp>::const_reference) const [with _Tp = const double; __gnu_cxx::new_allocator<_Tp>::const_pointer = const double*; __gnu_cxx::new_allocator<_Tp>::const_reference = const double&]’ cannot be overloaded
       address(const_reference __x) const _GLIBCXX_NOEXCEPT
       ^~~~~~~
/usr/include/c++/7/ext/new_allocator.h:89:7: error: with ‘_Tp* __gnu_cxx::new_allocator<_Tp>::address(__gnu_cxx::new_allocator<_Tp>::reference) const [with _Tp = const double; __gnu_cxx::new_allocator<_Tp>::pointer = const double*; __gnu_cxx::new_allocator<_Tp>::reference = const double&]’
       address(reference __x) const _GLIBCXX_NOEXCEPT
       ^~~~~~~
In file included from /home/thofmann/code/openrave/plugins/rplanners/rrt.h:19:0,
                 from /home/thofmann/code/openrave/plugins/rplanners/rplanners.cpp:18:
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h: In instantiation of ‘class VectorWrapper<const double>’:
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h:217:124:   required from here
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h:51:5: error: ‘VectorWrapper<T>::VectorWrapper(const T*, const T*) [with T = const double]’ cannot be overloaded
     VectorWrapper(const T* sourceArray, const T* sourceArrayEnd)
     ^~~~~~~~~~~~~
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h:44:5: error: with ‘VectorWrapper<T>::VectorWrapper(T*, T*) [with T = const double]’
     VectorWrapper(T* sourceArray, T* sourceArrayEnd)
     ^~~~~~~~~~~~~
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h: In member function ‘OpenRAVE::dReal SpatialTree<Node>::_ComputeDistance(const dReal*, const dReal*) const’:
/home/thofmann/code/openrave/plugins/rplanners/rplanners.h:217:124: error: no match for call to ‘(const boost::function<double(const std::vector<double>&, const std::vector<double>&)>) (VectorWrapper<const double>, VectorWrapper<double>)’
[... 208 lines cut ...]
-- Looking for dAllocateODEDataForThread - found
-- Detected ODE multi-threading extensions, but not sure if should enable them... define ODE_USE_MULTITHREAD variable to use them
-- Performing Test NEED_TRIINDEX
-- Performing Test NEED_TRIINDEX - Failed
-- ODE found, building oderave plugin, precision=SINGLE
-- Checking for module 'fcl'
--   Found fcl, version 0.5.0
-- Performing Test FCL_HAS_REPLACEOBJECT
-- Performing Test FCL_HAS_REPLACEOBJECT - Failed
-- Performing Test FCL_SUPPORT_BULK_UPDATE
-- Performing Test FCL_SUPPORT_BULK_UPDATE - Failed
-- Configuring done
CMake Warning (dev) at python/bindings/CMakeLists.txt:66 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "ANN" of target "pyANN_int" does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/thofmann/code/openrave/build
```

This is on Fedora 26 with GCC 7.2.1 20170915 (Red Hat 7.2.1-2) without any patches or custom build flags.

morxa, 2017-10-06:
The same error occurs with the 0.9.0 release. See the attached [cmake.log](https://github.com/rdiankov/openrave/files/1362615/cmake.log) and
[build.log](https://github.com/rdiankov/openrave/files/1362617/build.log).

I have the following relevant packages installed:
```
Installed Packages
OpenSceneGraph-devel.x86_64 3.4.0-9.fc26
OpenSceneGraph-qt-devel.x86_64 3.4.0-9.fc26
SoQt-devel.x86_64 1.5.0-21.fc26
ann-devel.x86_64 1.1.2-9.fc26
assimp-devel.x86_64 3.3.1-8.fc26
boost-devel.x86_64 1.63.0-9.fc26
boost-python.x86_64 1.63.0-9.fc26
bullet-devel.x86_64 2.83-3.fc26
cmake.x86_64 3.9.1-1.fc26
collada-dom-devel.x86_64 2.4.4-6.fc26
crlibm-devel.x86_64 1.0-0.14.beta4.fc26
dos2unix.x86_64 7.3.4-2.fc26
doxygen.x86_64 1:1.8.13-9.fc26
fparser-devel.x86_64 4.5.2-5.fc26
gettext.x86_64 0.19.8.1-9.fc26
gmp-devel.x86_64 1:6.1.2-4.fc26
graphviz.x86_64 2.40.1-4.fc26
lapack-devel.x86_64 3.6.1-4.fc26
libxml2-devel.x86_64 2.9.4-2.fc26
minizip-devel.x86_64 1.2.11-2.fc26
mpfi-devel.x86_64 1.5.1-10.fc26
mpfr-devel.x86_64 3.1.5-3.fc26
octave-devel.x86_64 6:4.2.1-3.fc26
ode-devel.x86_64 0.14-4.fc26
pcre-devel.x86_64 8.41-1.fc26
python2-devel.x86_64 2.7.13-12.fc26
qhull-devel.x86_64 2015.2-2.fc26
sympy.noarch 1.1.1-1.fc26
```

rdiankov, 2017-10-07:
thanks for the heads up. just pushed a fixed to master.

closed 2017-10-07 by rdiankov