rdiankov/openrave #537: [ikfast] Controlling number of solutions and enforcing joint limitations

hamzamerzic, 2017-12-07:
Hi. I am wondering if there is a way to control to number of solutions provided by ikfast and enforcing joint limitations? 
Currently I either get 0 or 8 solutions, and in case I get solutions, a lot of them fall out of the allowed joint range.

Thanks!

rdiankov, 2017-12-08:
At the moment, ikfast does not encode joint limits, so you'll have to check the joint limits once all the solutions have been computed.

closed 2017-12-08 by rdiankov

hamzamerzic, 2017-12-08:
Thanks for quick reply @rdiankov! I am also curious about increasing the number of solutions. Is that possible?

Ahrimanox, 2018-02-26:
Hi. I'm also interested about why does ikfast only returns 0, 4 or 8 joint solutions.
I haven't found any reasons or discussions about that !