rdiankov/openrave #543: GCC build fails due to not finding mpfr.h

jgvictores, 2018-01-08:
Build fails due to not finding mpfr.h
- Branch: `master` (b2818fb534cd8c4e64ea1dddea0f7c360ce5056d).
- Compiler/Distro: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.5) 5.4.0 20160609
- Error message:
    ```bash
    openrave/3rdparty/fparser-4.5.2/mpfr/MpfrFloat.cc:3:18: fatal error: mpfr.h: No such file or directory
    ```

Temporal workaround is to not use openrave version, and instead previously install package (inspired by [this](https://github.com/micheloosterhof/cowrie/issues/214#issuecomment-235654498)), via `sudo apt install libmpfr-dev`.

cielavenir, 2018-01-19:
actually I'm thinking of opposite; if libmpfr-dev is not installed, mpfr should not be enabled (by cmake).

jgvictores, 2018-01-19:
> actually I'm thinking of opposite; if libmpfr-dev is not installed, mpfr should not be enabled (by cmake).

Makes sense.