rdiankov/openrave #544: Failed on getting RGB image from qtcoin viewer

xiaoyuchhusky, 2018-01-12:
Hi everyone,

I am a new user of openrave. For my current project, I need get RGB image data in the openrave simulator. However, when I run the following scripts, I got the errors shown below.

from openravepy import *
import numpy, time
env = Environment() # create the environment
env.SetViewer('qtcoin') # start the viewer
env.Load('data/testwamcamera.env.xml')
env.GetSensors()
sensor = env.GetSensors()[0]
sensor.Configure(Sensor.ConfigureCommand.PowerOn)

Error:
In [9]: Coin warning in glxglue_context_create_software(): Couldn't create GLX context.
2018-01-11 20:46:17,429 openrave [WARN] [qtcoinviewer.cpp:3448 _GetCameraImage] offscreen renderer failed (check video driver), disabling
2018-01-11 20:46:17,658 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:17,888 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:18,119 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:18,350 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:18,581 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:18,811 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:19,042 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:19,273 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:19,503 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:19,732 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:19,960 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:20,190 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:20,419 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:20,648 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:20,879 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:21,111 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:21,341 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:21,570 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:21,800 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:22,029 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:22,260 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:22,490 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:22,721 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen
2018-01-11 20:46:22,951 openrave [WARN] [qtcoinviewer.cpp:3401 _GetCameraImage] cannot render offscreen

Does anyone know how to solve this? I am running openrave 0.9.0 under ubuntu 14.04 with graphics card  GeForce GTX 960 with nvidia-384 driver. Thank you very much!

cielavenir, 2018-01-12:
try to enable indirect GLX https://askubuntu.com/questions/745135/how-to-enable-indirect-glx-contexts-iglx-in-ubuntu-14-04-lts-with-nvidia-gfx

xiaoyuchhusky, 2018-01-19:
@cielavenir Thank you so much! It solves my problem.

closed 2018-01-19 by xiaoyuchhusky