rdiankov/openrave #547: Robot, loaded by .dae file, is not shown in 'qtcoin' viewer.

xiaoyuchhusky, 2018-01-19:
Hello everyone,

First, I already loaded a fetch robot in openrave by using "urdf" and "srdf" file, and the robot could be seen in the 'qtcoin' viewer. Then, I save the robot to a .dae file by:

`env.Save('fetch.dae',Environment.SelectionOptions.Robots)`

Then, I create an environment.xml, in which I have:
<Environment>
  <bkgndcolor>1 1 1</bkgndcolor>
  <camtrans>5.327509 -3.765386 5.225026</camtrans>
  <camrotationaxis>-0.916210 -0.318647 0.242947 138.160983</camrotationaxis>
  <Robot file="fetch.dae" name="fetch">
      <translation>0 0 0</translation>
      <jointvalues>0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.0 0.00 0.00</jointvalues>
  </Robot>
</Environment> 

Then, I load the 'environment.xml' file by running

`from openravepy import *`
`env = Environment()`
`env.SetViewer('qtcoin')`
`env.Load('/home/yuchen/projects/src/envs/environment.xml')`
`robot = env.GetRobots()[0]`
`robot.GetLinks()`

Then, I got the robot and it showed me all the robot links. However, I couldn't see anything in the viewer?

Does anyone know why and how to solve this? Thank you!

Best,

jgvictores, 2018-02-05:
As a small recommendation, try to compile OpenRAVE with openscenegraph (gets a more cartoon-ish look). It seems to work better with `.dae`. I haven't checked the code, buy my suspicion is that maybe openscenegraph's parser is used under the hood.

jgvictores, 2020-01-25:
Then again, looks like https://github.com/rdiankov/openrave/issues/435