rdiankov/openrave #551: Python modules do not build against Boost.Python>1.65.1

rmaronson, 2018-02-02:
Building with Boost.Python version 1.65.1 or greater is broken, as the `boost::python::numeric` namespace has been removed in favor of `boost::python::numpy`. See [version notes](http://www.boost.org/users/history/version_1_65_1.html).

marynelv, 2018-08-20:
Yup. I can confirm this. I was able to build against boost 1.58 (1.57 has some other issue as well).

morxa, 2018-12-23:
This also fixes #542.

lianghongzhuo, 2020-05-07:
Any plan to Migrate from boost::numeric to boost:numpy?

cielavenir, 2020-05-08:
we are migrating to pybind11. no document how to install it yet, but you could learn from https://github.com/cielavenir/mujin_recruiting/blob/pybind11/cookbooks/mujin/recipes/default.rb#L106

lianghongzhuo, 2020-05-08:
@cielavenir Can you successfully compile the master branch in your computer? If so,  what's your operating system and version? Thanks.

cielavenir, 2020-05-09:
@lianghongzhuo Debian stretch/buster, Ubuntu bionic/focal